RaspAP/raspap-webgui #863: Install script not working to install insiders version

zbchristian, 2021-04-11:
@billz The script at `https://install.raspap.com` does not work with `--insiders`, if no branch is defined.
Reason: RASPAP_LATEST is not set in this case.

Fixed `function _get_release ()`: 
```
function _get_release() {
    if [ "$insiders" == 1 ]; then
        repo="RaspAP/raspap-insiders"
        repo_common="RaspAP/raspap-webgui"
        readonly RASPAP_LATEST=$(curl -s "https://api.raspap.com/repos/RaspAP/raspap-insiders/releases/latest/" | grep -Po '"tag_name": "\K.*?(?=")' )
        readonly RASPAP_RELEASE="${RASPAP_INSIDERS_LATEST} Insiders"
    else
        readonly RASPAP_LATEST=$(curl -s "https://api.github.com/repos/$repo/releases/latest" | grep -Po '"tag_name": "\K.*?(?=")' )
        readonly RASPAP_RELEASE="${RASPAP_LATEST}"
    fi
}
```

billz, 2021-04-11:
@zbchristian see RaspAP/raspap-insiders/issues/20. This change reintroduces the bug. 
The issue is that forcing users to load `common.sh` from the private repo requires use of a personal access token passed with the `--token` switch, else the installer fails silently. 

Using the `--insiders` switch as-is presents no difficulties:

```
$ curl -sL https://install.raspap.com | bash -s -- --insiders

 888888ba                              .d888888   888888ba
 88     8b                            d8     88   88     8b
a88aaaa8P' .d8888b. .d8888b. 88d888b. 88aaaaa88a a88aaaa8P
 88    8b. 88    88 Y8ooooo. 88    88 88     88   88
 88     88 88.  .88       88 88.  .88 88     88   88
 dP     dP  88888P8  88888P  88Y888P  88     88   dP
                             88
                             dP      version 2.7.2 Insiders

The Quick Installer will guide you through a few easy steps

RaspAP Install: Configure installation
Detected OS: Raspbian GNU/Linux 10 (buster)
Using GitHub repository: RaspAP/raspap-insiders 2.7.2 branch
```

or maybe I'm missing something?

billz, 2021-04-11:
Worth mentioning that this will _not_ install updates to `common.sh` in a feature branch (eg., `feature/zbchristian clients`) within the private repo. I believe this is the issue here. What I would suggest instead is that use of `--token` would enable setting `$repo_common` to a private repo that requires authentication.

Requiring a token at install time from insiders users is a non-starter.

zbchristian, 2021-04-11:
@billz Ah yes, you are right. I was (again) confused, that `common.sh` is pulled from the webgui version.
I was caught by this, because I passed the `--insiders` together with a branch name. This is only possible using `--repo RaspAP/raspap-insiders`

billz, 2021-04-11:
I can see how combining those switches would lead to this. The `--insiders` switch was added purely to simplify the install/upgrade from this repo.

closed 2021-12-07 by billz