RaspAP/raspap-webgui #870: The ui sometimes shows the wrong interface

moham96, 2021-04-19:
## Checklist

- [x ] This is a bug report
- [x ] I have searched this repository for existing issues
- [ x] I checked the FAQ and official documentation before creating this issue
- [ x] I have read and understand the issue reporting guidelines 

## Bug description

when changing hotspot settings in the ui, sometimes the ui shows the wrong interface, for example after selecting wlan0 as the interface and refreshing the ui still shows (uap0) as the interface (or even worse some times it shows eth0) although by checking the file `/etc/raspap/hostapd.ini` i can see that the settings were saved, but the ui doesn't reload the settings apparently. 
## Your environment
1. Operating System: RPi OS 32-bit  
2. Hardware and version: RPi 3 b 
3. RaspAP version: RaspAP v2.6.4 
4. Followed the project prerequisites? Yes 
5. RaspAP Quick Install or Manual setup? Quick install 
6. Using default configuration? No 
7. Simultaneous AP and managed mode? Yes
8. Onboard wireless chipset or external adapter? Onboard 
9. Other software or services running with RaspAP? none

## Steps to reproduce

- install raspap using the quick install
-  set up ap-managed mode
- try to change the interface or the password a few times, you'll notice that the interface doesn't update from the ini file
## Screenshots

## Additional context

billz, 2021-04-19:
Understanding [how AP-STA works](https://docs.raspap.com/ap-sta/) will help explain this. The virtual interface `uap0` doesn't exist until the AP-STA config is saved and the raspap.service is restarted.

I wouldn't classify this as a bug, but probably a UX issue.

connected201, 2021-04-20:
problem exist! i confirm.

closed 2021-04-20 by billz