RaspAP/raspap-webgui #871: how to connect 3g usb dongle to share internet over wlan?

connected201, 2021-04-20:
hi, i have installed RaspAP v2.6.4
i have modem that work in system:
[image]
[image]
but here i cannot see my 3g modem:
[image]
how to make to receive internet from 3g modem and share to access point?

connected201, 2021-04-20:
UPD, after enabling Bridged AP mode, i lost access to my raspberry form eth0 raspberry ip was 192.168.100.105, and after enabling, i cannot connect to my web interface!

billz, 2021-04-21:
Two things: 1) please use the issue template; it's there for a reason. 2) mobile client support is in development for the [Insiders edition](https://docs.raspap.com/insiders/) of RaspAP.

billz, 2021-04-21:
> after enabling Bridged AP mode, i lost access to my raspberry

https://docs.raspap.com/bridged/#accessing-the-web-interface

closed 2021-04-21 by billz