RaspAP/raspap-webgui #882: Openvpn not working with CyberGhost

TheLonelyTomato, 2021-04-29:
Hey guys, i hope  you will have enough information for an answer.
Great job and keep it up  👍
PS : i'm new to RSP so answers for idiots please 😉

Describe the bug

Openvpn not working with CyberGhost

1- I enter the user name and password
2- Add .ovpn file from CyberGhost website
3- Save settings see picture :
[image]
4- Start OpenVPN  = "Attempting to start OpenVPN" but nothing red light
 

Your environment

1. Operating System: Raspbian desktop Kernel version: 5.10
2. Hardware and version: RPI 3 b+
3. RaspAP version: 08164f4
4. Followed the project prerequisites? Yes
5. RaspAP Quick Install or Manual setup? Quick install
6. Using default configuration? Yes
7. Simultaneous AP and managed mode? No idea
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP?  No

closed 2021-04-29 by billz

TheLonelyTomato, 2021-04-29:
What did i do wrong ??

billz, 2021-04-29:
https://docs.raspap.com/issues/