RaspAP/raspap-webgui #893: Installation not working

yoshiy999, 2021-05-06:
## Checklist

- [ x] This is a bug report
- [ x] I have searched this repository for existing issues
- [ x] I checked the FAQ and official documentation before creating this issue
- [ x] I have read and understand the issue reporting guidelines 

## Bug description

## Your environment
1. Operating System: RPi OS 32-bit Lite
2. Hardware and version: RPi 3
3. RaspAP version: latest
4. Followed the project prerequisites? Yes
5. RaspAP Quick Install or Manual setup? Quick install
6. Using default configuration? Yes
7. Simultaneous AP and managed mode? 
8. Onboard wireless chipset or external adapter? External (LB-Link 150mbs)
9. Other software or services running with RaspAP? No

## Steps to reproduce
Hi, I've been trying to install the Raspap both normal and insiders but I'm not able to do it. Every time I run the install command, it reads me the bash instead of running it. 
Is there an issue with the raspap bash install script?
I've done about 15 fresh install. I was able to run the bash script once (on the 5th attempt) and then tried to upgrade to insiders but it wouldn't work. It was said some folders couldn't be moved.  I did the exact same process every time for the fresh install but still not able to install.

Thanks

yoshiy999, 2021-05-06:
I think for some reasons there are 2 vertical bar that can be used on my keyboard and the one I was using messed up the command line

closed 2021-05-06 by yoshiy999

billz, 2021-05-06:
Yes, the output from curl is piped to bash with the `|` symbol. Not sure about the double bar on your keyboard, maybe a logical OR symbol? In any case, copy-pasting the command from here is usually easiest https://raspap.com/#quick