RaspAP/raspap-webgui #897: RaspAP and Pi-hole

tefracky, 2021-05-12:
## Checklist

- [X] This is a bug report
- [X] I have searched this repository for existing issues
- [X] I checked the FAQ and official documentation before creating this issue
- [X] I have read and understand the issue reporting guidelines 

## Bug description

I want to use RaspAP as AP parallel to PiHole. I searched a lot, but I didn't find a sufficient solution for the dnsmasq problem. The FAQ says that I have either use RaspAP as Bridged AP, using two different devices or delete PiHole. Since this is not what I want I would kindly ask, if there is any solution to use RaspAP as AP with DHCP parallel to PiHole.

## Your environment
1. Operating System: RPi OS 32-bit 
2. Hardware and version: RPi 4 
3. RaspAP version: 2.6.5 
4. Followed the project prerequisites? Yes 
5. RaspAP Quick Install or Manual setup? Quick Install 
6. Using default configuration? Yes 
7. Simultaneous AP and managed mode? No 
8. Onboard wireless chipset or external adapter? Onboard 
9. Other software or services running with RaspAP?  Pi-Hole

## Steps to reproduce

1. Install Pi-Hole
2. Install RaspAP

## Screenshots

## Additional context

billz, 2021-05-12:
> if there is any solution to use RaspAP as AP with DHCP parallel to PiHole

Short answer (as described in the FAQ): no. 
If you don't want to use a second device you can run RaspAP or Pi-hole in a Docker container.

tefracky, 2021-05-12:
Thanks! It is sad, but true. I tried to run RaspAP in a Docker container, but the conflict with port 53 still exist.

billz, 2021-05-12:
You might want to raise an issue re: port 53 (DNS) on the raspap-docker repo.

We've made no attempts to integrate these projects, so there's no support for this. Suggest you try Pi-hole's Docker container and/or search their issues. For now, the best alternative is to use AdGuard Home if you want to use RaspAP on the same device, or run Pi-hole on a second one.

closed 2021-06-06 by billz