RaspAP/raspap-webgui #912: DHCP don't set the ip for dispositives

crudo0-arch, 2021-05-25:
## Checklist

- [ ] This is a bug report
- [ ] I have searched this repository for existing issues
- [ ] I checked the FAQ and official documentation before creating this issue
- [ ] I have read and understand the issue reporting guidelines 

## Bug description
I have a RPI whit sim hat (wwan0) and an alfa usb antenna (wlan1). I don't know why it dosen't work, smartphone don't get the ip. Please help me... my configuration on the gui in the image below, and my log here:

from 10.19.124.244
May 23 17:44:36 dnsmasq[14261]: cached content-autofill.googleapis.com is 172.217.21.74
May 23 17:48:16 dnsmasq[14261]: query[A] tyrqwkpdw from 10.19.124.244
May 23 17:48:16 dnsmasq[14261]: config tyrqwkpdw is NODATA-IPv4
May 23 17:48:16 dnsmasq[14261]: query[A] ojerrzitrug from 10.19.124.244
May 23 17:48:16 dnsmasq[14261]: config ojerrzitrug is NODATA-IPv4
May 23 17:48:16 dnsmasq[14261]: query[A] lsqerooosdmft from 10.19.124.244
May 23 17:48:16 dnsmasq[14261]: config lsqerooosdmft is NODATA-IPv4
May 23 17:48:18 dnsmasq[14261]: exiting on receipt of SIGTERM
May 23 17:48:25 dnsmasq[17504]: started, version 2.80 cachesize 150
May 23 17:48:25 dnsmasq[17504]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
May 23 17:48:25 dnsmasq[17504]: warning: interface wlan0 does not currently exist
May 23 17:48:25 dnsmasq[17504]: warning: interface eth0 does not currently exist
May 23 17:48:25 dnsmasq[17504]: warning: interface wlan1 does not currently exist
May 23 17:48:25 dnsmasq-dhcp[17504]: DHCP, IP range 10.3.141.50 -- 10.3.141.255, lease time 12h
May 23 17:48:25 dnsmasq-dhcp[17504]: DHCP, IP range 10.3.141.50 -- 10.3.141.255, lease time 12h
May 23 17:48:25 dnsmasq-dhcp[17504]: DHCP, IP range 10.3.141.50 -- 10.3.141.255, lease time 12h
May 23 17:48:25 dnsmasq-dhcp[17504]: DHCP, IP range 10.3.141.50 -- 10.3.141.255, lease time 12h
May 23 17:48:25 dnsmasq[17504]: reading /run/dnsmasq/resolv.conf
May 23 17:48:25 dnsmasq[17504]: using nameserver 151.5.216.25#53
May 23 17:48:25 dnsmasq[17504]: using nameserver 151.5.216.225#53
May 23 17:48:25 dnsmasq[17504]: read /etc/hosts - 5 addresses
May 23 17:48:28 dnsmasq[17504]: read /etc/raspap/adblock/hostnames.txt - 249820 addresses
May 23 17:48:30 dnsmasq[17504]: read /etc/raspap/adblock/hostnames.txt - 249820 addresses
May 23 17:48:30 dnsmasq[17504]: query[A] content-autofill.googleapis.com from 10.19.124.244
May 23 17:48:30 dnsmasq[17504]: forwarded content-autofill.googleapis.com to 151.5.216.25
May 23 17:48:30 dnsmasq[17504]: forwarded content-autofill.googleapis.com to 151.5.216.225
May 23 17:48:30 dnsmasq[17504]: query[A] uqhthgxtr from 10.19.124.244
May 23 17:48:30 dnsmasq[17504]: config uqhthgxtr is NODATA-IPv4
May 23 17:48:30 dnsmasq[17504]: query[A] vhwzosjqmukjpu from 10.19.124.244
May 23 17:48:30 dnsmasq[17504]: config vhwzosjqmukjpu is NODATA-IPv4
May 23 17:48:30 dnsmasq[17504]: query[A] ofogmxtov from 10.19.124.244
May 23 17:48:30 dnsmasq[17504]: config ofogmxtov is NODATA-IPv4
May 23 17:48:30 dnsmasq[17504]: reply content-autofill.googleapis.com is 142.250.180.138
May 23 17:48:38 dnsmasq[17504]: exiting on receipt of SIGTERM
May 23 17:48:48 dnsmasq[17775]: started, version 2.80 cachesize 150
May 23 17:48:48 dnsmasq[17775]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
May 23 17:48:48 dnsmasq[17775]: warning: interface wlan0 does not currently exist
May 23 17:48:48 dnsmasq[17775]: warning: interface eth0 does not currently exist
May 23 17:48:48 dnsmasq-dhcp[17775]: DHCP, IP range 10.3.141.50 -- 10.3.141.255, lease time 12h
May 23 17:48:48 dnsmasq-dhcp[17775]: DHCP, IP range 10.3.141.50 -- 10.3.141.255, lease time 12h
May 23 17:48:48 dnsmasq-dhcp[17775]: DHCP, IP range 10.3.141.50 -- 10.3.141.255, lease time 12h
May 23 17:48:48 dnsmasq-dhcp[17775]: DHCP, IP range 10.3.141.50 -- 10.3.141.255, lease time 12h
May 23 17:48:48 dnsmasq[17775]: reading /run/dnsmasq/resolv.conf
May 23 17:48:48 dnsmasq[17775]: using nameserver 151.5.216.25#53
May 23 17:48:48 dnsmasq[17775]: using nameserver 151.5.216.225#53
May 23 17:48:48 dnsmasq[17775]: read /etc/hosts - 5 addresses
May 23 17:48:52 dnsmasq[17775]: read /etc/raspap/adblock/hostnames.txt - 249820 addresses
May 23 17:48:53 dnsmasq[17775]: read /etc/raspap/adblock/hostnames.txt - 249820 addresses
May 23 17:48:53 dnsmasq[17775]: query[A] gkfnupi from 10.19.124.244
May 23 17:48:53 dnsmasq[17775]: config gkfnupi is NODATA-IPv4
May 23 17:48:53 dnsmasq[17775]: query[A] awkrmfonw from 10.19.124.244
May 23 17:48:53 dnsmasq[17775]: config awkrmfonw is NODATA-IPv4
May 23 17:48:53 dnsmasq[17775]: query[A] spphunnq from 10.19.124.244
May 23 17:48:53 dnsmasq[17775]: config spphunnq is NODATA-IPv4
May 23 17:48:53 dnsmasq[17775]: query[A] content-autofill.googleapis.com from 10.19.124.244
May 23 17:48:53 dnsmasq[17775]: forwarded content-autofill.googleapis.com to 151.5.216.25
May 23 17:48:53 dnsmasq[17775]: forwarded content-autofill.googleapis.com to 151.5.216.225
May 23 17:48:53 dnsmasq[17775]: query[A] content-autofill.googleapis.com from 10.19.124.244
May 23 17:48:53 dnsmasq[17775]: query[A] cnmntgnhc from 10.19.124.244
[... 92 lines cut ...]
May 23 17:52:50 dnsmasq[17775]: forwarded one.one.one.one to 151.5.216.225
May 23 17:52:50 dnsmasq[17775]: reply one.one.one.one is 1.0.0.1
May 23 17:52:50 dnsmasq[17775]: reply one.one.one.one is 1.1.1.1
May 23 17:52:50 dnsmasq[17775]: reply one.one.one.one is 2606:4700:4700::1001
May 23 17:52:50 dnsmasq[17775]: reply one.one.one.one is 2606:4700:4700::1111
May 23 17:52:50 dnsmasq[17775]: query[PTR] 1.0.0.1.in-addr.arpa from 10.19.124.244
May 23 17:52:50 dnsmasq[17775]: forwarded 1.0.0.1.in-addr.arpa to 151.5.216.225
May 23 17:52:50 dnsmasq[17775]: reply 1.0.0.1 is one.one.one.one
[image]

## Your environment
1. Operating System: RASBIAN 
2. Hardware and version: 
4. Followed the project prerequisites? 
5. RaspAP Quick Install or Manual setup? 
6. Using default configuration? 
7. Simultaneous AP and managed mode? 
8. Onboard wireless chipset or external adapter? 
9. Other software or services running with RaspAP? 

## Steps to reproduce

## Screenshots

## Additional context

billz, 2021-05-25:
https://docs.raspap.com/issues/

closed 2021-05-25 by billz

crudo0-arch, 2021-05-25:
There aren't solutions in that, link. I have search how to solve on FAQ and on the docs. 
Why have you closed?