RaspAP/raspap-webgui #914: No Internet connection

tefracky, 2021-05-28:
## Checklist

- [X] This is a bug report
- [X] I have searched this repository for existing issues
- [X] I checked the FAQ and official documentation before creating this issue
- [X] I have read and understand the issue reporting guidelines 

## Bug description

I instaled RaspAP and in principle, the AP is working. I can connect to it, the DHCP server works, but I can't connect to the internet at all. I searched a lot and it seems, the IPv4 forwarding isn't workng correctly. I am not familiar with interfaces, forwardings etc., but I am really unsure, what I have to do exactly and what log files/outputs you need. I am very sorry for wasting your time, but I really don't know what I have to do.

## Your environment
1. Operating System: RPi OS 32-bit 
2. Hardware and version: RPi 4 
3. RaspAP version: 2.6.5 
4. Followed the project prerequisites? Yes 
5. RaspAP Quick Install or Manual setup? Quick Install 
6. Using default configuration? Yes 
7. Simultaneous AP and managed mode? No 
8. Onboard wireless chipset or external adapter? Onboard 
9. Other software or services running with RaspAP?  Pi-Hole

## Steps to reproduce

1. Install RaspAP
2. Activate an AP
3. Connect to the AP
4. Try to connect to the internet

## Screenshots

## Additional context

billz, 2021-05-28:
> Other software or services running with RaspAP? Pi-Hole

https://docs.raspap.com/faq/#pihole

tefracky, 2021-05-29:
> > Other software or services running with RaspAP? Pi-Hole
> 
> https://docs.raspap.com/faq/#pihole

I just removed pihole and use another device for PiHole.

billz, 2021-06-06:
Start with a clean install per the project prerequisites. Closing this, reopen if your issue persists

closed 2021-06-06 by billz