RaspAP/raspap-webgui #917: wpa_supplicant scan_results with unicode names

glaszig, 2021-05-29:
given the following scan result:

```
$ sudo wpa_cli -i wlan1 scan_results
bssid / frequency / signal level / flags / ssid
00:12:23:34:56:3a	2462	-59	[WPA2-PSK-CCMP][ESS]	Foo Bar
18:12:23:34:56:40	2447	-69	[WPA2-PSK-CCMP][WPS][ESS]	Foo Bar 2
c0:12:23:34:56:a6	2427	-83	[WPA-PSK-CCMP][WPA2-PSK-CCMP][ESS]	Maria Cec\xc3\xadlia
58:12:23:34:56:e8	2462	-85	[WPA-PSK-CCMP][WPA2-PSK-CCMP][WPS][ESS]	@LaLaLa
```

does anybody of you folks have an idea how to decode that `\xc3\xad` into the propre unicode sign?
i fiddled around with several php encoding/decoding functions but didn't arrive anywhere.

billz, 2021-05-29:
The `\x` prefix indicates a hexadecimal byte. These sequences can be echoed with a [double quoted string](https://www.php.net/manual/en/language.types.string.php#language.types.string.syntax.double). 

```
echo("Maria Cec\xc3\xadlia");
Maria Cecília
```

assuming the default utf-8 page encoding.

glaszig, 2021-05-29:
yeah. doing it this way in the console it works. it does not work, though, with `echo $ssid` in `templates/wifi_stations.php`.

o.0

glaszig, 2021-05-30:
for reference: https://raspberrypi.stackexchange.com/questions/41069/how-do-i-configure-wpa-supplicant-with-default-iphone-ssid#55556

scan result is being parsed with preg_match. so php's double-quote-escape-sequence-handling does not kick in.

glaszig, 2021-05-30:
proof of concept:

`unicode-ssid.stdout`

```
bssid / frequency / signal level / flags / ssid
00:12:23:34:56:3a	2462	-59	[WPA2-PSK-CCMP][ESS]	Foo Bar
18:12:23:34:56:40	2447	-69	[WPA2-PSK-CCMP][WPS][ESS]	Foo Bar 2
c0:12:23:34:56:a6	2427	-83	[WPA-PSK-CCMP][WPA2-PSK-CCMP][ESS]	Maria Cec\xc3\xadlia
58:12:23:34:56:e8	2462	-85	[WPA-PSK-CCMP][WPA2-PSK-CCMP][WPS][ESS]	@LaLaLa
```

`unicode-ssid.php`

```php
<?php

$s = null;
exec('cat unicode-ssid.stdout', $s);
array_shift($s);

$n = $s[2];
$n = preg_split("/[\t]+/", $n);

$ssid = $n[4];
$ssid = 'Maria Cec\xc3\xadlia';

echo "Original: $ssid\n";

function convertHex($input) {
	return hex2bin($input[1]);
};
$ssid = preg_replace_callback('/\\\x(..)/u', 'convertHex', $ssid);
echo "Fixed-up: $ssid\n";
```

result:

```
$ php unicode-ssid.php
Original: Maria Cec\xc3\xadlia
Fixed-up: Maria Cecília
```

PR #918 (2021-05-31): interpret hex sequences in ssid station names
resolves #917

before:
[image]

after:
[image]
commit 92e154fa (2021-05-31): interpret hex sequences in ssid station names

resolves #917
closed 2021-05-31 by billz