RaspAP/raspap-webgui #929: Hotspot Update Causing Duplicate Entries in dhcpcd.conf File

ryanklein, 2021-06-09:
When updating the Hotspot configuring and changing the interface or wireless mode it generates new, and sometimes incorrect, entires in the dhcpcd.conf file.

```pi@:~ $ cat /etc/dhcpcd.conf 
# RaspAP default configuration
hostname
clientid
persistent
option rapid_commit
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes
option ntp_servers
require dhcp_server_identifier
slaac private
nohook lookup-hostname

# RaspAP wlan0 configuration
interface wlan0
static ip_address=10.3.141.1/24
static routers=10.3.141.1
static domain_name_server=1.1.1.1 8.8.8.8

# RaspAP wlan0 configuration
interface wlan0
static ip_address=/INF
static routers=
static domain_name_server=

denyinterfaces wwan0
denyinterfaces wwan0
# RaspAP wlan0 configuration
interface wlan0
static ip_address=/INF
static routers=
static domain_name_server=

pi@:~ $ 
```

billz, 2021-06-09:
Please resubmit with the issue template https://docs.raspap.com/issues/

closed 2021-06-09 by billz

ryanklein, 2021-06-09:
@billz I’ve updated the bug report, can this be reopened or is another bug report needed?

billz, 2021-06-10:
@ryanklein steps to reproduce this are particularly useful.