RaspAP/raspap-webgui #954: High peak memory usage

ninok, 2021-07-08:
## Checklist

- [ x ] This is a bug report
- [ x ] I have searched this repository for existing issues
- [ x ] I checked the FAQ and official documentation before creating this issue
- [ x ] I have read and understand the issue reporting guidelines 

## Bug description
I recently installed RaspAP on my RaspberryPi 4 (2GB) and everything works nicely out of the box.

The only thing that happens from time to time is that the memory consumption peaks to above 50%.

Top shows there are many `dnsmasq` instances running when this happens.

## Your environment
1. Operating System: RPi OS 32-bit Lite
2. Hardware and version:  RPi 4 2GB
3. RaspAP version: 2.6.6
4. Followed the project prerequisites? Yes
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? Yes
7. Simultaneous AP and managed mode? No, AP only
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP? Internet connection via Huawei LTE stick

## Steps to reproduce
Install with quick install and take a look at memory consumption from time to time

ninok, 2021-07-08:
Screenshot of processes sorted by memory
[image]

billz, 2021-07-09:
Those values are highly unusual for a typical RaspAP install. 

For comparison, a 4GB RPi4 running RaspAP continuously for 28 days shows 16% memory usage. dnsmasq is very lightweight and a single instance consumes little system resources. RaspAP does not spawn multiple instances of dnsmasq. 

```
$ free -m | awk '/Mem:/ { total=$2 ; used=$3 } END { print used/total*100}'
16.6276
```

```
$ ps aux | grep dnsmasq
dnsmasq   1032  0.0  8.3  81568 73020 ?        S    Jul01   4:02 /usr/sbin/dnsmasq -x /run/dnsmasq/dnsmasq.pid -u dnsmasq -r /run/dnsmasq/resolv.conf -7 /etc/dnsmasq.d,.dpkg-dist,.dpkg-old,.dpkg-new --local-service
```

```
top - 15:46:42 up 28 days,  6:28,  2 users,  load average: 0.00, 0.00, 0.00
Tasks: 133 total,   1 running, 132 sleeping,   0 stopped,   0 zombie
%Cpu(s):   0.0/0.2     0[                                                                                ]
MiB Mem : 27.3/854.5    [|||||||||||||||||||||||||||                                                     ]
MiB Swap:  0.0/100.0    [                                                                                ]

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                               
 8180 pi        20   0   10332   3020   2476 R   0.3   0.3   0:04.28 top                                                                                                                                   
 8289 root      20   0       0      0      0 I   0.3   0.0   0:00.01 kworker/3:1-events_power_efficient                                                                                                    
    1 root      20   0   34836   8324   6504 S   0.0   1.0   0:24.45 systemd                                                                                                                               
    2 root      20   0       0      0      0 S   0.0   0.0   0:01.12 kthreadd                                                                                                                              
    3 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 rcu_gp                                                                                                                                
    4 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 rcu_par_gp                                                                                                                            
    8 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 mm_percpu_wq                                                                                                                          
    9 root      20   0       0      0      0 S   0.0   0.0   0:00.00 rcu_tasks_rude_                                                                                                                       
   10 root      20   0       0      0      0 S   0.0   0.0   0:00.00 rcu_tasks_trace                                                                                                                       
   11 root      20   0       0      0      0 S   0.0   0.0   0:07.52 ksoftirqd/0                                                                                                                           
   12 root      20   0       0      0      0 I   0.0   0.0   0:31.47 rcu_sched                                                                                                                             
   13 root      rt   0       0      0      0 S   0.0   0.0   0:00.63 migration/0                                                                                                                           
   14 root      20   0       0      0      0 S   0.0   0.0   0:00.00 cpuhp/0                                                                                                                               
   15 root      20   0       0      0      0 S   0.0   0.0   0:00.00 cpuhp/1                                                                                                                               
   16 root      rt   0       0      0      0 S   0.0   0.0   0:00.31 migration/1                                                                                                                           
   17 root      20   0       0      0      0 S   0.0   0.0   0:00.46 ksoftirqd/1                                                                                                                           
   20 root      20   0       0      0      0 S   0.0   0.0   0:00.00 cpuhp/2                                                                                                                               
   21 root      rt   0       0      0      0 S   0.0   0.0   0:00.31 migration/2                                                                                                                           
   22 root      20   0       0      0      0 S   0.0   0.0   0:03.30 ksoftirqd/2                                                                                                                           
   25 root      20   0       0      0      0 S   0.0   0.0   0:00.00 cpuhp/3                                                                                                                               
[... cut ...]
   67 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 kthrotld                                                                                                                              
   68 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 iscsi_eh                                                                                                                              
   69 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 iscsi_destroy                                                                                                                              
```

ninok, 2021-07-11:
Thanks @billz. Do you have any idea why this could happen. I'm using the RaspAP adblock feature that uses dnsmasq afaik. Could that lead to such high memory consumption?

Tstef21, 2021-10-05:
On raspberry pi 3b i have with DNS dnsmasq adblocking a high load. Without it it's a small load.

My pihole with over 100k blocklist have also a Big loaded memory.

closed 2021-11-27 by billz