RaspAP/raspap-webgui #968: OpenVPN at startup

manubissanchez, 2021-07-24:
## Checklist

- [x] This is a bug report
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description
OpenVPN don't start properly at boot. I found this : https://github.com/RaspAP/raspap-webgui/issues/437
But I have this message when it tries to connect just after the boot : 
`Broadcast message from root@hostname (Sat 2021-07-24 22:08:12 CEST):

Password entry required for 'Enter Private Key Password:' (PID 964).
Please enter password with the systemd-tty-ask-password-agent tool:`

I found this for fix it : [https://forums.openvpn.net/viewtopic.php?t=27419](url)
It works when I stop then start openvpn from the admin page but not at the startup.

## Your environment
1. Operating System: RPi OS 32-bit Lite 
2. Hardware and version: RPi Zero W 
3. RaspAP version: 2.6.6 
4. Followed the project prerequisites? Yes
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? Yes 
7. Simultaneous AP and managed mode? No 
8. Onboard wireless chipset or external adapter? onboard 
9. Other software or services running with RaspAP? No

## Steps to reproduce

## Screenshots

## Additional context

billz, 2021-07-25:
RaspAP's OpenVPN implementation does not support passphrase protected private keys. 

The private key pass uses weak encryption so you are better off stripping it from the key. There are many examples online of how to do this with openssl.

manubissanchez, 2021-07-25:
Thank you. I don't really understand your answer but I will learn. I use pivpn on the server side. If I use "pivpn -a nopass" to generate a ovpn file, it doesn't work. Do you have a tutorial for this please ?

billz, 2021-07-26:
You should be able to do something like this:

`openssl rsa -in my-encrypted.key -out my-decrypted.key`

...or search issues at pivpn, or use your workaround.

Tstef21, 2021-10-05:
You can upload the ovpn file direct to /etc/openvpn and IT works again.
After this restart and See the Status on the raspap web interface.. VPN is up

closed 2023-12-13 by github-actions[bot]