RaspAP/raspap-webgui #973: Special instructions for Pi Zero w are 404?

dubbledex, 2021-07-28:
## Checklist

- [ ] This is a bug report
- [ ] I have searched this repository for existing issues
- [ x] I checked the FAQ and official documentation before creating this issue
- [ ] I have read and understand the issue reporting guidelines 

## Bug description

Not a bug as such, but the link to the special instructions for Pi Zero W goes to a 404 page. I had a look on the documentation and there was info in the FAQ, but it takes you to the same 404 page.. What do you need to do different for a Pi Zero?
Thanks :)

## Your environment
1. Operating System: **ENTER HERE** 
2. Hardware and version: 
3. RaspAP version: 
4. Followed the project prerequisites? 
5. RaspAP Quick Install or Manual setup? 
6. Using default configuration? 
7. Simultaneous AP and managed mode? 
8. Onboard wireless chipset or external adapter? 
9. Other software or services running with RaspAP? 

## Steps to reproduce

## Screenshots

## Additional context

billz, 2021-07-29:
FAQ updated. You could also use the docs search function :)

closed 2021-07-29 by billz

dingausmwald, 2021-07-30:
Erm.. readme.md still refers to https://docs.raspap.com/ap-sta.html, which is 404. In addition, when i follow the instructions (which seem to be misleading) i cant activate WiFi client AP mode. Installation detects Raspbian, which is wrong. Its Raspberry Pi OS lite, like advised here: https://docs.raspap.com/ap-sta/

Following this guide, i start with:

1. SD Card flashing Raspberry PI OS (32-bit) lte.
2. copy wpa_supplicant to boot
3. copy ssh file to boot
4. insert sd card and start the pi
5. connect to pi via ssh
6. Following the project prerequisites by this link https://docs.raspap.com/#quick-start (which doenst mention anything about prerequisites, so i assume the steps are:
6.1 `sudo apt-get update`
6.2 `sudo apt-get full-upgrade`
6.3 `sudo reboot`
6.4 `sudo raspi-config` and change WiFi country in localisation options (hope the other menues can be change too - locale, timezone, keyboard - does nothin when no keyboard is attached)
7. Start installer with `curl -sL https://install.raspap.com | bash`
8. like on the picture, the installer detects the pi and does adding sudoers, simlink wpa_supplicant, unmasks and enables hostapd_service and so on

but when logging in, opening hostpot > advanced, WIFI client AP mode is greyed out / unavailable. Everything else seem to be working. Wifi link and so on. Tried 10 times. What am i doing wrong?!

It is written: 

> Note: The WiFi client AP mode option will be disabled, or "greyed out", until a wireless client is configured.

But from my opinion this is made by step 2, right?
Seems like the whole rpi zero documentation needs an update. Couldnt find anything in the issues.