RaspAP/raspap-webgui #983: Raspap server crashes when changing wifi to 5Ga

Nezara, 2021-08-09:
## Checklist

- [x ] This is a bug report
- [x ] I have searched this repository for existing issues
- [ x] I checked the FAQ and official documentation before creating this issue
- [ x] I have read and understand the issue reporting guidelines 

## Bug description

Hostapd fails when changing port from 80 to 8080 on 802.11a setting. 

## Your environment
1. Operating System: RPi OS 32
2. Hardware and version: Pi4 4GB
3. RaspAP version: v2.6.6
4. Followed the project prerequisites? Yes?
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? Mostly, modified port number only
7. Simultaneous AP and managed mode? No
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP?  N/A 
## Steps to reproduce

I changed the port from 80 to 8080 in advanced settings so I could host a local website (as per the official RaspAP documentation). Changing the hotspot from default to 5G (from 802.11g to 802.11a), saving and then restarting hotspot causes hostapd to fail.

## Screenshots

## Additional context

billz, 2021-08-09:
Unable to repro with the info provided. Post relevant log entries indicating a failed state for these services (please do _not_ dump your entire log). 
https://docs.raspap.com/ap-basics/#troubleshooting

Nezara, 2021-08-09:
I think I found the issue.

I grabbed the hostapd logs which indicated that the channel was not compatible with the wifi frequency being used, eg channel 1 is not compatible with 802.11a. Unfortunately RaspAP only lets you select from channels 1-11, all of which are incompatible with the 5hz frequency. Manually editing the hostapd.conf file to select wifi "a" and channel "36" resolves the issue.  The channel now displays as blank on the RaspAP UI. 

I might suggest an update on the UI there to open more channels depending on the wifi type selected. 

Cheers

commit 3145b605 (2021-08-09): Legacy 802.11a UI support. Resolves #983
closed 2021-08-09 by billz