RaspAP/raspap-webgui #994: Wrong variable for DNS-Serverlist

SephGER, 2021-08-25:
https://github.com/RaspAP/raspap-webgui/blob/48feef88ffa06979acc9773fe12d8b2fdd6e5e4f/templates/dhcp/advanced.php#L20

The list of DNS-Servers doesn't load for me. After changing  `$upstreamServers` to `$conf['server']` it works.

billz, 2021-08-25:
This var is, or should be, set here:

https://github.com/RaspAP/raspap-webgui/blob/061d01cbd56411adbe52a313ff8d04e2bfa893bc/ajax/networking/get_netcfg.php#L22

You can help our devs diagnose this by using the issue template, as described below. Otherwise we’re building a test case with no steps to repro, etc.
https://docs.raspap.com/issues/

billz, 2021-09-07:
@SephGER thanks for reporting. I've mentioned you in the commit msg

commit c2410400 (2021-09-07): Pass $upstreamServers to template, resolves #994 thx @SephGER
closed 2021-09-07 by billz