RaspAP/raspap-webgui #1003: Cannot Connect to webgui (10.3.141.1) on Pi 3B+

s1ashtv, 2021-09-11:
## Checklist

- [x] This is a bug report
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

After installing using the quick installer and rebooting my Pi, when i go to chromium, and type 10.3.141.1 which is what the documentation said, it didnt allow me to connect and displays an error ERR_CONNECTION_REFUSED on my chromium. Tried the solutions provided in issue #588 but still dont work for me.
## Your environment
1. Operating System: **Raspberry Pi Os 32-Bit** 
2. Hardware and version:  RPi 3B+
3. RaspAP version:  v2.6.6
4. Followed the project prerequisites?  Yes
5. RaspAP Quick Install or Manual setup?  Quick Install
6. Using default configuration?  Yes
7. Simultaneous AP and managed mode?  Yes
8. Onboard wireless chipset or external adapter?  Onboard wireless chipset
9. Other software or services running with RaspAP? 

## Steps to reproduce

## Screenshots

[image]

## Additional context

billz, 2021-09-12:
https://docs.raspap.com/faq/#access

s1ashtv, 2021-09-14:
Thanks for the FAQ, upon checking, i think it is on the lighttpd part, and upon typing systemctl status lighttpd, here is what i found:

> ● lighttpd.service - Lighttpd Daemon
   Loaded: loaded (/lib/systemd/system/lighttpd.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2021-09-14 21:59:06 PST; 45s ago
  Process: 1928 ExecStartPre=/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf (code=exited, s
  Process: 1933 ExecStart=/usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf (code=exited, statu
 Main PID: 1933 (code=exited, status=255/EXCEPTION)

> Sep 14 21:59:06 raspberrypi systemd[1]: lighttpd.service: Service RestartSec=100ms expired, schedu

> Sep 14 21:59:06 raspberrypi systemd[1]: lighttpd.service: Scheduled restart job, restart counter i

> Sep 14 21:59:06 raspberrypi systemd[1]: Stopped Lighttpd Daemon.

> Sep 14 21:59:06 raspberrypi systemd[1]: lighttpd.service: Start request repeated too quickly.

> Sep 14 21:59:06 raspberrypi systemd[1]: lighttpd.service: Failed with result 'exit-code'.

> Sep 14 21:59:06 raspberrypi systemd[1]: Failed to start Lighttpd Daemon.

is there a workaround regarding this or a fix? Thanks

billz, 2021-09-14:
There are no issues with lighttpd failing to start on a clean installs RaspAP. Follow the project prerequisites.
Other tips here https://docs.raspap.com/issues/#issue-policy

closed 2021-09-14 by billz