RaspAP/raspap-webgui #1031: Invalid CSRF TOKEN

rdmitry0911, 2021-10-16:
## Checklist

- [x] This is a bug report
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

Hi guys, I know that in the past issue 377 was discussed and closed as resolved, however, on my system I'm getting this error every time I want to save something using web interface.

## Your environment
1. Operating System: **Raspbian GNU/Linux 10 (buster)** 
2. Hardware and version: Raspberry Pi 3 Model B Plus Rev 1.3 
3. RaspAP version: RaspAP v2.6.7 
4. Followed the project prerequisites? yes 
5. RaspAP Quick Install or Manual setup? Quick Install 
6. Using default configuration? yes 
7. Simultaneous AP and managed mode? no 
8. Onboard wireless chipset or external adapter? Onboard 
9. Other software or services running with RaspAP? asterisk PBX

## Steps to reproduce

any attempt to save modified information using web ui leads to this error "Invalid CSRF token"
## Screenshots

## Additional context

billz, 2021-10-16:
> enable Asterisk's built-in micro HTTP server

Apparently this package installs its own micro HTTP service. I suspect this is your problem. 

Suggest you run RaspAP's web service, lighttpd, on an alternate port. See [this FAQ](https://docs.raspap.com/faq/#can-i-configure-an-alternate-port-for-raspaps-web-service) for advice.

rdmitry0911, 2021-10-16:
> Apparently this package installs its own micro HTTP service. I suspect this is your problem.
>
> Suggest you run RaspAP's web service, lighttpd, on an alternate port. See [this [...]

I moved lighttpd server to port 8081. The problem persists.

billz, 2021-10-16:
Did you dump $_SERVER variables? Have you checked issues on Asterisk's repo?

This is not an issue on clean installs of RaspAP. I can only suggest workarounds for your integration.
Bear in mind this is not a support forum for third-party software.

rdmitry0911, 2021-10-16:
> Did you dump $_SERVER variables? Have you checked issues on Asterisk's repo?
> 
> This is not an issue on clean installs of RaspAP. I can only suggest workarounds for your integration. Bear in mind this is not a support forum for third-party software.

I've just checked. In $_SERVER there is no HTTP_X_CSRF_TOKEN variable. How can I put it in there? I have no other web server running at the moment, however in the past there was a running web server.