RaspAP/raspap-webgui #1045: RaspAP stops after ~30 minutes

IAW9927, 2021-11-06:
## Checklist

- [ x] This is a bug report
- [ x] I have searched this repository for existing issues
- [ x] I checked the FAQ and official documentation before creating this issue
- [ x] I have read and understand the issue reporting guidelines 

## Bug description

## Your environment
1. Operating System: Raspberry Pi OS 32-bit 
2. Hardware and version: RPi 4B
3. RaspAP version: v2.6.7
4. Followed the project prerequisites? yes 
5. RaspAP Quick Install or Manual setup? quick install
6. Using default configuration? No
7. Simultaneous AP and managed mode? yes
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP? nginx, kiwix-serve

## Steps to reproduce

Keep the hotspot enabled for a few hours
## Screenshots

## Additional context

IAW9927, 2021-11-06:
Bug description is: The RaspAP service dis-assigns the IP 10.3.141.1 and stops the AP.

billz, 2021-11-06:
As with your previous issue #1043. Please respect this project's issue policy

closed 2021-11-06 by billz

IAW9927, 2021-11-06:
The log of hostapd is:
Configuration file: /etc/hostapd/hostapd.conf
Could not read interface uap0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=uap0 disabled_11b_rates=0
Could not read interface uap0 flags: No such device
nl80211 driver initialization failed.
uap0: interface state UNINITIALIZED->DISABLED
uap0: AP-DISABLED 
uap0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface uap0 wasn't started
Configuration file: /etc/hostapd/hostapd.conf
Could not read interface uap0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=uap0 disabled_11b_rates=0
Could not read interface uap0 flags: No such device
nl80211 driver initialization failed.
uap0: interface state UNINITIALIZED->DISABLED
uap0: AP-DISABLED 
uap0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface uap0 wasn't started
Configuration file: /etc/hostapd/hostapd.conf
Could not read interface uap0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=uap0 disabled_11b_rates=0
Could not read interface uap0 flags: No such device
nl80211 driver initialization failed.
uap0: interface state UNINITIALIZED->DISABLED
uap0: AP-DISABLED 
uap0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface uap0 wasn't started
Configuration file: /etc/hostapd/hostapd.conf
Could not read interface uap0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=uap0 disabled_11b_rates=0
Could not read interface uap0 flags: No such device
nl80211 driver initialization failed.
uap0: interface state UNINITIALIZED->DISABLED
uap0: AP-DISABLED 
uap0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface uap0 wasn't started
Configuration file: /etc/hostapd/hostapd.conf
Could not read interface uap0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=uap0 disabled_11b_rates=0
Could not read interface uap0 flags: No such device
nl80211 driver initialization failed.
uap0: interface state UNINITIALIZED->DISABLED
uap0: AP-DISABLED 
uap0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface uap0 wasn't started
Configuration file: /etc/hostapd/hostapd.conf
Could not read interface uap0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=uap0 disabled_11b_rates=0
Could not read interface uap0 flags: No such device
nl80211 driver initialization failed.
uap0: interface state UNINITIALIZED->DISABLED
uap0: AP-DISABLED 
uap0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface uap0 wasn't started
Configuration file: /etc/hostapd/hostapd.conf
uap0: interface state UNINITIALIZED->COUNTRY_UPDATE
uap0: Could not connect to kernel driver
Using interface uap0 with hwaddr e4:5f:01:16:d1:ea and ssid "Knowledge-OpenWiFi-2.4G"
uap0: interface state COUNTRY_UPDATE->ENABLED
uap0: AP-ENABLED 
uap0: STA 60:5b:b4:22:51:8f IEEE 802.11: associated
uap0: AP-STA-CONNECTED 60:5b:b4:22:51:8f
uap0: STA 60:5b:b4:22:51:8f RADIUS: starting accounting session 9D194E8AC88FB554
uap0: STA 60:5b:b4:22:51:8f IEEE 802.11: disassociated
uap0: AP-STA-DISCONNECTED 60:5b:b4:22:51:8f
uap0: STA d0:01:cf:01:d0:01 IEEE 802.11: disassociated
uap0: STA 60:5b:b4:22:51:8f IEEE 802.11: disassociated