RaspAP/raspap-webgui #1046: Moving /tmp/dnsmasq.log to /var/log/raspap.log

andresayang, 2021-11-07:
## Checklist

- [X] This is a bug report / Improvement easy to implement
- [X] I have searched this repository for existing issues
- [X] I checked the FAQ and official documentation before creating this issue
- [X] I have read and understand the issue reporting guidelines 

## Bug description

dnsmasq service do not restart with systemctl status message: "cannot open log /tmp/dnsmasq.log: Permission denied" 

## Your environment
1. Operating System: **Debian 11 (Well i know it is not officially supported)** 
2. Hardware and version: RPi 3B+, RPi 4 
3. RaspAP version: 2.6.7
4. Followed the project prerequisites? 
5. RaspAP Quick Install or Manual setup? Quick Install (from procedure on website)
6. Using default configuration? No
7. Simultaneous AP and managed mode? No
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP? 
No, I already found a closed same subject, I only indicate an easy way to bypass all dnsmasq log troubles 
## Steps to reproduce

Simply reboot on debian 11, dnsmasq do not restart due to "/tmp/dnsmasq.log: Permission denied"
If you only change the default install log-facility to /var/log/raspap.log it will:
- Allow log to be accessible at reboot
- Bypass other "dnsmasq" log to co-exists
## Screenshots

## Additional context

matzrm, 2021-11-08:
also here, with debian 11 raspap doesn't start because dnsmasq cannot open or read the log.

movergan, 2021-11-09:
Hit it yesterday too

matzrm, 2021-11-09:
@movergan I solved the issue removing dnsmasq, and re-compiling it from source (find it here: git clone http://thekelleys.org.uk/git/dnsmasq.git ). Then you need to unmask the dnsmasq service and restore the default raspap conf files in /etc/dnsmasq .

billz, 2021-11-09:
Debian 11 was just released and is currently unsupported.

Bear in mind that extending support to new distros also requires acceptance testing on all currently supported ones.

andresayang, 2021-11-13:
> Debian 11 was just released and is currently unsupported.
> 
> Bear in mind that extending support to new distros also requires acceptance testing on all currently supported ones.

Hello, Yes we all know debian 11 is not supported: The thing is that you will have to support it soon or later (well I hope so).
However, 
I did a totally different install with apache2 on a third machine (on debian 12 - testing as i wanted to use gnome 40) so now I can not anymore "claim for mods". I'll do them myself.

Cheers

closed 2021-11-13 by billz

billz, 2021-11-15:
Latest release has support for Debian 11. Thanks for reporting

https://github.com/RaspAP/raspap-webgui/releases/tag/2.6.8

andresayang, 2021-11-18:
> Latest release has support for Debian 11. Thanks for reporting
> 
> https://github.com/RaspAP/raspap-webgui/releases/tag/2.6.8

Thanks a lot