RaspAP/raspap-webgui #1051: Feature request: Improve support for open networks

billz, 2021-11-13:
PR #1050 removes the password requirement for the WiFi client configuration UI. This facilitates connections to open wireless networks. 

While open WiFi networks may present a security risk, the point was made in #1039 that this would enable users of RaspAP to potentially use connectivity from networks managed by captive portals.

shinigami-78, 2021-11-23:
Thank you for enabling this. If we're using the Insider edition, is this change included in the current release? (I installed it just tonight).

billz, 2021-11-23:
The pull request is still open. I usually like to get feedback prior to merging, but since this is an Insider's feature request I'll merge it and draft a new release. This will also include PRs by @zbchristian, along with a Debian 11-related breaking change to vnStat which we use to generate traffic statistics. So quite a few updates.

I'll make an announcement when it's done. Thanks :)

closed 2021-11-29 by billz