RaspAP/raspap-webgui #1060: No traffic stats report - RaspAP 2.6.9 on RPi OS [Bullseye]

ivnmad, 2021-11-17:
## Checklist

- [ ] This is a bug report
- [ ] I have searched this repository for existing issues
- [ ] I checked the FAQ and official documentation before creating this issue
- [ ] I have read and understand the issue reporting guidelines 

## Bug description

## Your environment
1. Operating System: RPi OS 32-bit Lite [Bullseye]
2. Hardware and version: RPi 3B+
3. RaspAP version: 2.6.9
4. Followed the project prerequisites? Yes
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? Yes
7. Simultaneous AP and managed mode? No
8. Onboard wireless chipset or external adapter? Onboard

## Steps to reproduce
New RRPi OS [Bullseye] installation
New RaspAP 2.6.9 installation

## Screenshots

[image]

[image]

## Additional context

vnstat shows traffic that was generated following the RaspAP installation, but RaspAP reports are empty.

billz, 2021-11-18:
The updated version of the vnStat package in Debian 11 changed a column name from 'hours' to 'hour'. This broke parsing of stats. Thanks for reporting.

closed 2021-11-18 by billz

billz, 2021-11-18:
Dashboard stats with the fix applied.
The 'undefined' period occurred while debugging.

[image]

Miniontoby, 2021-11-21:
Hey,

Since this fix my rpi didn't work anymore with the graph! 
I had to change the `hour` to `hours` again to let it work... (after seeing it 1 hour later)

- Miniontoby

ivnmad, 2021-11-21:
> Hey,
> 
> Since this fix my rpi didn't work anymore with the graph! I had to change the `hour` to `hours` again to let it work... (after seeing it 1 hour later)

Since the fix I can see the hourly graphs, but they are incorrect, because it doesn't manage to detect the date/hour correctly. A lot of entries show "undetermined". The daily data usage graphs and tables are empty.
I reported this on a separate bug report:
https://github.com/RaspAP/raspap-webgui/issues/1064

billz, 2021-11-21:
Debian 11 installs an updated version of vnStat (2.6) that breaks RaspAP's stats implementation. This is due to a non-backwards compatible change to vnstat's JSON output.

For this reason, RaspAP's latest release is bumped from 2.6.9 to 2.7.0.

mcbmcb, 2022-01-22:
> Hey,
> 
> Since this fix my rpi didn't work anymore with the graph! I had to change the `hour` to `hours` again to let it work... (after seeing it 1 hour later)
> 
> * Miniontoby

I don't understand in what file I can change "hours" to "hour" (or vice-versa ?) I'm on raspberry Buster and don't see any stats on the dashboard. (RaspAP v2.8.3 ; vnStat 1.18 by Teemu Toivola <tst at iki dot fi>)

Guray00, 2022-01-30:
Same problem here. What should we do?

billz, 2022-01-30:
Disregard the comments above. See the [release notes](https://github.com/RaspAP/raspap-webgui/releases/tag/2.7.0).

commit a426823a (2021-11-18): Bugfix: vnstat column name change. Resolves #1060