RaspAP/raspap-webgui #1068: Data Usage - KB instead of MB

ivnmad, 2021-11-27:
## Checklist

- [ ] This is a bug report
- [ ] I have searched this repository for existing issues
- [ ] I checked the FAQ and official documentation before creating this issue
- [ ] I have read and understand the issue reporting guidelines 

## Bug description
Data Usage is in KB and not MB.
E.g. from attached screenshot, on 27th Nov it shows 22983 sent traffic, but that is 22983 KB and not 22983 MB.
We should either change the key/legend from MB to KB, or add a dot to make 22983 into 22.983, or remove the last 3 digits and make 22983 into 23 [if we roundup]

## Your environment
1. Operating System: RPi OS 32-bit Lite [Bullseye]
2. Hardware and version:  RPi 3B+/
3. RaspAP version: 2.7.1
4. Followed the project prerequisites? Yes
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? Yes
7. Simultaneous AP and managed mode? No
8. Onboard wireless chipset or external adapter? Onboard

## Steps to reproduce
Install RaspAP, connected a device and generated some traffic.

## Screenshots

[image]

paul908, 2022-01-07:
As a quick and dirty refactoring - change the phrases in messages.po in https://github.com/RaspAP/raspap-webgui/blob/master/locale/en_US/LC_MESSAGES/messages.po and in all locales.

fsufyan, 2022-01-21:
Same issue appears for me on RaspAP latest version 2.8.3.
[image]

Maybe we can enhance this and automatically adjust the transfer unit according to traffic size, for example: **KB**, **MB**, **GB**.

commit 2b3b0eb6 (2022-04-24): fix #1068 data size units
closed 2022-04-24 by billz