RaspAP/raspap-webgui #1072: WiFi client AP mode greyed out even when household network is "Connected" in "WiFi Client" menu

johnhess, 2021-11-30:
## Checklist

- [x] This is a bug report
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

After using a freshly imaged SD card and following the project prerequisites on my Raspberry Pi 3B+ then connecting to a wireless network, the "WiFi client AP mode" slider remains greyed out.

## Your environment
1. Operating System: RPi OS 32-bit Lite (bullseye) 
2. Hardware and version: RPi 3B+
3. RaspAP version: 2.7.0 and 2.7.1 
4. Followed the project prerequisites? Yes 
5. RaspAP Quick Install or Manual setup? Quick Install 
6. Using default configuration? Yes 
7. Simultaneous AP and managed mode? Yes (or at least that's the goal)
8. Onboard wireless chipset or external adapter? Onboard 
9. Other software or services running with RaspAP? No.

## Steps to reproduce

I've followed the instructions and pre-reqs.  The only possible deviation is I'm using Raspberry Pi Imager to set a hostname, an SSH password and provide the household WiFi SSID/Password so the RPi can be reached over the network via SSH to continue the configuration.

[image]

Immediately after the quick installer is run (using defaults), I can successfully reach the RaspAP UI.  In "WiFi client", I see my household network under "Connected" (it'd have to be, given that I can reach it from my laptop which is also connected to the household network).

`iwgetid wlan0` confirms this connection

```
pi@butterbox:~ $ iwgetid wlan0
wlan0     ESSID:"My household SSID"
```

Yet, *Hotspot>Advanced>WiFi client AP mode* remains greyed out.

[image]

Of course, restarting at this point turns on the AP but leaves the Pi disconnected from the household network

## Additional context
At no point in the process above was the RPi connected over ethernet. 

I'm trying to develop a script to install RaspAP (and later some other software) with the goal of using on a Pi Zero 2 W I've ordered.  I am able to connect and debug the 3B+ over ethernet if that helps.

Happy to re-attempt from a fresh install and pull logs, etc.

ThatOneRoadie, 2021-12-08:
I had the same issue with a recent install. It looks like the quick installer didn't detect that Eth0 was not connected, and never set up the uap0 interface, so Wifi Client AP Mode remains greyed out, and there's no uap0 interface anywhere.

billz, 2021-12-08:
(re)configure the wireless client using the UI. AP-STA mode will then be available.
the steps are documented here https://docs.raspap.com/ap-sta/

@ThatOneRoadie the installer does not create the uap0 interface.

closed 2021-12-08 by billz