RaspAP/raspap-webgui #1076: quick installer problem with iptables: Chain 'MASQUERADE' does not exist

kiai72, 2021-12-05:
## Checklist

- [x] This is a bug report
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

## Your environment
1. Operating System: RPI OS 32 bit lite (buster)
2. Hardware and version: RPi Zero W
3. RaspAP version: 2.7.1
4. Followed the project prerequisites? Yes, it's only to upgrad, right ?
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? Yes
7. Simultaneous AP and managed mode? I don't even get that far
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP? No

## Steps to reproduce
I Just start the quick installer via ssh and get this error:
...
RaspAP Install: Configuring networking
Enabling IP forwarding
net.ipv4.ip_forward = 1
Restarting procps (via systemctl): procps.service.
Checking iptables rules
grep: /etc/iptables/rules.v4: No such file or directory
Adding rule: -t nat -A POSTROUTING -j MASQUERADE
iptables v1.8.2 (nf_tables): Chain 'MASQUERADE' does not exist
Try `iptables -h' or 'iptables --help' for more information.
[ ? error ]  Unable to execute iptables
grep: /etc/iptables/rules.v4: No such file or directory
Adding rule: -t nat -A POSTROUTING -s 192.168.50.0/24 ! -d 192.168.50.0/24 -j MASQUERADE
iptables v1.8.2 (nf_tables): Chain 'MASQUERADE' does not exist
Try `iptables -h' or 'iptables --help' for more information.
[ ? error ]  Unable to execute iptables
Persisting IP tables rules

## Screenshots
here's my complete log: https://pastebin.com/RwkgxMdL

## Additional context
informations from neofetch:
  /oooo++//ooooo:  ooooo+//+ooooo.    OS: Raspbian GNU/Linux 10 (buster) armv6l
  `+ooooooo:-:oo-  +o+::/ooooooo:     Host: Raspberry Pi Zero W Rev 1.1
   `:oooooooo+``    `.oooooooo+-      Kernel: 4.14.83-aufs
     `:++ooo/.        :+ooo+/.`       Uptime: 1 hour, 40 mins
        ...`  `.----.` ``..           Packages: 548 (dpkg)
     .::::-``:::::::::.`-:::-`        Shell: bash 5.0.3
    -:::-`   .:::::::-`  `-:::-       Terminal: /dev/pts/1
   `::.  `.--.`  `` `.---.``.::`      CPU: BCM2835 (1) @ 1.100GHz
       .::::::::`  -::::::::` `       Memory: 46MiB / 368MiB

billz, 2021-12-05:
Your iptables rules are not in a standard configuration.
If you have already defined iptables rules you need to adapt this project's rules to your config.
See the manual install steps for more info.

You should also follow the project prerequisites and install the latest release of RPi OS.

Clean RPi OS (bullseye) install ... 
```
RaspAP Install: Configuring networking
Enabling IP forwarding
net.ipv4.ip_forward = 1
Restarting procps (via systemctl): procps.service.
Checking iptables rules
Adding rule: -t nat -A POSTROUTING -j MASQUERADE
Adding rule: -t nat -A POSTROUTING -s 192.168.50.0/24 ! -d 192.168.50.0/24 -j MASQUERADE
Persisting IP tables rules
```

closed 2021-12-05 by billz

kiai72, 2021-12-07:
this is a clean buster image. I'm using berryboot, but is this relevant ?
Buster is supported as well, isn't it ?
I started from scratch. Did "sudo update-alternatives --config iptables" (legacy) before - and this time iptables worked.

moaipool, 2021-12-08:
Per the project reqs:

> Begin with a clean install of the latest release of a supported Linux distribution.

billz, 2021-12-08:
Per the project reqs:

> Begin with a clean install of the latest release of a supported Linux distribution.

nomasteryoda, 2022-05-06:
AND more importantly, try to install it from the Console - i.e. your monitor - instead of across the network. I learned the hard way and had to do a clean install. 
Other tips: set the hostname before trying to install raspap.