RaspAP/raspap-webgui #1081: Data Usage Never Populated

shinigami-78, 2021-12-11:
## Checklist

- [x] This is a bug report
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

Since upgrading to the Insider edition v2.8.0, the data usage dashboard always shows zero (0) traffic. This happens on both the "Dashboard" and the "Data usage" page. I am using RaspAP with both the onboard wireless chipset (wlan0) and an external adapter (wlan1). In my set up, wlan0 creates a hotspot and wlan1 connects as a client to a wireless network so that traffic flows in through wlan1 and out to the internet via wlan0. In the prior version v2.7.9, the data usage was correctly reported; this issue appeared only after I upgraded. 

## Your environment
1. Operating System: RPi OS 32-bit Lite 
2. Hardware and version: RPi 4 
3. RaspAP version: RaspAP v2.8.0 
4. Followed the project prerequisites? Yes 
5. RaspAP Quick Install or Manual setup? Quick Install 
6. Using default configuration? Yes 
7. Simultaneous AP and managed mode? No 
8. Onboard wireless chipset or external adapter? Both 
9. Other software or services running with RaspAP? 

## Steps to reproduce

## Screenshots

[image]

[image]

## Additional context

billz, 2021-12-11:
See the release notes re: Debian 11 (bullseye) and vnStat. 
https://github.com/RaspAP/raspap-webgui/releases/tag/2.7.0

shinigami-78, 2021-12-11:
I'm still on Buster though. I've not upgraded to Bullseye to avoid breaking RaspAP :-)

$ cat /etc/os-release

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

billz, 2021-12-11:
Sorry the release notes are not more clear. Bullseye installs a new major release version of vnStat:

```
apt-cache policy vnstat
vnstat:
  Installed: (none)
  Candidate: 2.6-3
```

Buster uses 1.18-2. vnStat's JSON data output between these versions is non-backwards compatible. ie., RaspAP under Bullseye was unable to parse vnStat's JSON. Users wasted no time raising alerts about their broken stats. We make every effort to try to make RaspAP backwards compatible with previous OS releases, to a reasonable degree. In this case there's no workaround except to upgrade to Bullseye.

We don't get to choose which packages are upgraded in these OS releases, unfortunately.

shinigami-78, 2021-12-11:
Ah I read the release note as "don't upgrade because things will break" but it's actually the opposite? 😀 The solution to the problem
I see is to upgrade to Bullseye?

billz, 2021-12-13:
Yes, that's correct.

billz, 2021-12-15:
Marking this as resolved. Feel free to reopen if this issue persists, thanks.

closed 2021-12-15 by billz