RaspAP/raspap-webgui #1084: Reset to default IP wlan0 after change of SSID name

marcogenna, 2021-12-16:
After a default setup I changed wlan0 IP to 192.168.101.1/24 and DHPC service starts from 192.168.101.90 ends to 192.168.101.150
After chancing SSID and rebooting wlan0 IP is back to default 10.3.141.1

## Checklist

- [x] This is a bug report
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

After a default setup I change wlan0 IP to 192.168.101.1/24 and DHPC service starts from 192.168.101.90 ends to 192.168.101.150
Make a reboot and the wlan0 IP is 192.168.101.1
After changing SSID and rebooting wlan0 IP is back to default 10.3.141.1 and DHCP remains with 192.168.101.90 to 192.168.101.150, when you connect to WiFi, no IP is assigned, dnsmasq and dhcpcd are alive, to come back need to connect to lan0 e cange again wlan0 IP to the same class of DHCP server.

## Your environment
1. Operating System:  RPi OS 32-bit Lite 
2. Hardware and version: RPi 3B+ 
3. RaspAP version: 2.7 
4. Followed the project prerequisites? Yes 
5. RaspAP Quick Install or Manual setup? Quick install
6. Using default configuration? No
7. Simultaneous AP and managed mode? No 
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP? 2 custom daemons working on FS C++/Python

## Steps to reproduce

## Screenshots

## Additional context

billz, 2021-12-17:
Unable to repro with info provided (steps to reproduce are missing).
Verify your changes to `/etc/dhcpcd.conf` before and after rebooting. Disable any other services that may be writing to this config.

closed 2021-12-17 by billz

marcogenna, 2021-12-17:
I updated	informations, is it possible to reopen the bug?
Thanks Marco

billz, 2021-12-17:
The wlan0 static IP was changed to 10.3.141.11. It is not reset to default after changing it, including after reboots.

```
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.3.141.11  netmask 255.255.255.0  broadcast 10.3.141.255
```

RaspAP writes to `/etc/dhcpcd.conf` when you click 'Save settings' and gets out of the way.
If this file is being modified by something else, it's your responsibility to trace it.