RaspAP/raspap-webgui #1091: Disconnection of the AP and "AP-STA-POSSIBLE-PSK-MISMATCH" error

p0lycarpio, 2022-01-01:
## Checklist

- [x] This is a bug report
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description
When using my access point with any terminal, after about 5 minutes the devices are disconnected. In the hostapd logs, I can read:
`wlan0: AP-STA-POSSIBLE-PSK-MISMATCH e6:36:05:cf:c8:56`
`wlan0: STA e6:36:05:cf:c8:56 IEEE 802.11: disassociated`

## Your environment
1. Operating System: **Raspbian Bullseye 32-bit Lite** 
2. Hardware and version: Raspberry Pi 4B 8GB
3. RaspAP version: 2.8.0 and 2.7.1
4. Followed the project prerequisites? Yes 
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? SSID, password, wireless mode and channel changed
7. Simultaneous AP and managed mode? No 
8. Onboard wireless chipset or external adapter? Onboard 
9. Other software or services running with RaspAP? Yes, Docker. I followed the recommendations on [FAQ page](https://docs.raspap.com/faq/#what-breaks-raspap-when-docker-is-installed-on-the-same-system-and-how-i-can-fix-it).

## Steps to reproduce

1. Install RPi OS 11 32-bit Lite
2. Follow quick-start installation steps
3. Change SSID, PSK, wireless mode and channel
4. Restart

## Screenshots

None

## Additional context
I don't think there is a conflict with Docker because stopping at startup and at runtime I still have the same problem.

qaqak, 2022-01-04:
Something like that happens with me, except it's that the `wlan0` adapter goes kaput

p0lycarpio, 2022-01-04:
After search, I think I have found what is causing the problem. I'm guessing the [project cockpit](https://github.com/cockpit-project/cockpit) related packages may interfere with `wlan0`. After uninstalling **cockpit** relative packages everything seems to work :)
Should we warn users of a possible conflict?

billz, 2022-01-04:
@p0lycarpio good to know, thanks. I looked into that error and couldn't find much on it. 
I'm unfamiliar with that software package, but think it's worth a mention in the FAQ.

qaqak, 2022-01-05:
I know this is the wrong place to post my issue, but I can only do it here. I'm sorry for doing it, and I'm not sorry for what you devs are about to see
You STUPID CARELESS DEVS have deleted all of my issues without notice!!!!!!
Why?!??!!?

qaqak, 2022-01-05:
AND YOU EVEN DELETED MY COMMENT!!!!

closed 2023-12-13 by github-actions[bot]