RaspAP/raspap-webgui #1095: Changing hotspot SSID or password reproducibly messes up working AP-STA mode setup

T0bi79, 2022-01-04:
## Checklist

- [x] This is a bug report
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

I made several fresh installs on current bullseye lite on Pi 3B+ (and partially Pi 400). Closely followed the installation guideline and **https://docs.raspap.com/ap-sta/** 
(despite the fact that Hotspot > Advanced requires setting a country code before the WiFi client AP mode will be ungreyed - so it seems the WIFI country already set by raspi-config is not respected here) which results in a WORKING setup in WiFi client AP mode.
Even after reboot this setup keeps working, while the hotspot is still using its default SSID and password.

Once I edit either the hotspots SSID or password, the hotspot won't start any more.
After a reboot, the RaspAP service daemon always fails to start the hotspot ("Failed to start Access point and authentication server for Wi-Fi and Ethernet."):
```
...
[  OK  ] Stopped Access point and authentication server for Wi-Fi and Ethernet.
         Starting Access point and authentication server for Wi-Fi and Ethernet...
[FAILED] Failed to start Access point and authentication server for Wi-Fi and Ethernet.
See 'systemctl status hostapd.service' for details.
[  OK  ] Started DHCP Client Daemon.
[  OK  ] Stopped Access point and authentication server for Wi-Fi and Ethernet.
         Starting Access point and authentication server for Wi-Fi and Ethernet...
[FAILED] Failed to start Access point and authentication server for Wi-Fi and Ethernet.
See 'systemctl status hostapd.service' for details.
         Starting dnsmasq - A lightweight DHCP and caching DNS server...
[  OK  ] Started dnsmasq - A lightweight DHCP and caching DNS server.
[  OK  ] Stopped Access point and authentication server for Wi-Fi and Ethernet.
         Starting Access point and authentication server for Wi-Fi and Ethernet...
[FAILED] Failed to start Access point and authentication server for Wi-Fi and Ethernet.
See 'systemctl status hostapd.service' for details.
[  OK  ] Finished RaspAP Service Daemon.
```
So after like 8 tries (I did not count exactly) the service daemon gives up and closes.
Before, in the working setup, this step always succeeded after 3 tries.

This is reproducible. I have a working image in the state with the default SSID and password...

## Your environment
1. Operating System: RPi OS 32-bit Lite
2. Hardware and version: RPi 3B+, (also 400)
3. RaspAP version: 2.8.1
4. Followed the project prerequisites? Yes
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? Yes
7. Simultaneous AP and managed mode? Yes
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP? No

## Steps to reproduce
This is more or less all stated above. Fresh install, following **https://docs.raspap.com/ap-sta/**  (no Ethernet, just Wifi), activating managed mode. This setup works even after reboots. Changing the hotspots default SSID or password afterwards will damage the setup.

## Screenshots
n/a

## Additional context
n/a

billz, 2022-01-04:
The error thrown by the daemon is a clue... `See 'systemctl status hostapd.service' for details.`
Refer to https://docs.raspap.com/ap-basics/#troubleshooting

closed 2022-01-04 by billz

T0bi79, 2022-01-04:
I already looked into' systemctl status hostapd.service', did not seem helpful:
```
● hostapd.service - Access point and authentication server for Wi-Fi and Ethernet
     Loaded: loaded (/lib/systemd/system/hostapd.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Tue 2022-01-04 12:51:28 CET; 745ms ago
       Docs: man:hostapd(8)
    Process: 1308 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, status=1/FAILURE)
        CPU: 30ms
```

More helpful seems the hostapd log in the web interface.
It tells
```
Configuration file: /etc/hostapd/hostapd.conf
Could not read interface uap0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=uap0 disabled_11b_rates=0
Could not read interface uap0 flags: No such device
nl80211 driver initialization failed.
uap0: interface state UNINITIALIZED->DISABLED
uap0: AP-DISABLED 
uap0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface uap0 wasn't started
```

This is trange, because all this uap0 stuff worked before changing ssid or password. It seems to break the previosly working uap0 configuration...

T0bi79, 2022-01-04:
Found a workaround after yet another reinstall.

Preparation: Fresh install as above. WiFi client AP mode functional with default credentials.
--> Backing up the disk image in this state.

**Test #1:** Booting up again. Changing `wpa_passphrase` and `ssid` manually by editing `/etc/hostapd/hostapd.conf'` Rebooting. Setup still functional with new SSID and password

Rolling back to the image from above and booting up again,  WiFi client AP mode is again functional with default credentials.

**Test #2:** Changing SSID and passphrase via the web interface. Restarting the hotspot using the button --> disfunctional. Restarting the pi --> still disfunctional. Same errors than already listed in previous posts.

**Conclusion:** Something breaks a functional AP mode hotspot when editing its settings using the web interface. Manually editing the `hostapd.conf` keeps the hotspot functional.

billz, 2022-01-04:
If I understand this correctly, the SSID and/or PSK is being changed while AP-STA mode is operating? This isn't clear in your report.

T0bi79, 2022-01-04:
Yes. As I wrote, I followed the tuturial at https://docs.raspap.com/ap-sta/ where the install script is used without active ethernet connection so that it omits a concluding reboot. In the final steps of the tutorial, the (yet inactive) hotspot is configured to AP-STA mode and started. This is where the tutorial ends and which works so far. Even across reboots this setup stays functional.

In such a system running in (functional) AP-STA mode I want to edit the SSID and the PSK. If I edit hostapd.conf manually (on the running system), save the changes and reboot, it still works. New SSID and PSK are functional. If I use the webinterface instead for the same changes (also while AP-STA is already in operation) the hotspot won't start correctly any more, neither by using the restart button nor by rebooting the system...

I found this whole thing very fragile in my firsts tests when Ethernet was still active. Several times I did not get AP-STA to work at all. That's why I finally followed the tutorial very closely with absolutely no Ethernet involved, So I did not yet try to adjust SSID and PSK already along with the initial configuration of the hotspot. But trying to change it afterwards definitely ends up in the same kind of messed-up config...

billz, 2022-01-04:
The hotspot (hostapd) must be stopped when in AP-STA mode before making changes to it.
The reason is indicated in your error logs: `uap0` is a virtual interface that is added to your wireless client (STA) interface to create the AP. Stop the hotspot, make your changes, then restart it. Problem solved.

T0bi79, 2022-01-04:
> The hotspot (hostapd) must be stopped when in AP-STA mode before making changes to it.

Ah, I did not expect that because the `WiFi client AP mode` button was also grayed out when the system is in a state that disallows its use.

>  Stop the hotspot, make your changes, then restart it. Problem solved.

I just tested it in strong expectation of a successful result. I restored my bootstrap image (working AP-STA config, default SSID/PSK), booted the pi (hotspot visible on my phone), stopped the hotspot via web interface (hotspot disappeared on my phone), changed SSID and PSK, saved it and restarted the hotspot. However, same error. Hotspot does not reappear on my phone (even after rebooting) and hostapd log still complains about "no such device" uap0...

I'll give this a second try once the image is restored again...

T0bi79, 2022-01-04:
Second try, same result.

Stopping the hotspot before changing SSID&PSK does not help.

Here's the last entry from the hostapd log (again...)

```
Configuration file: /etc/hostapd/hostapd.conf
Could not read interface uap0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=uap0 disabled_11b_rates=0
Could not read interface uap0 flags: No such device
nl80211 driver initialization failed.
uap0: interface state UNINITIALIZED->DISABLED
uap0: AP-DISABLED 
uap0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface uap0 wasn't started
```

billz, 2022-01-04:
Unable to repro. If you're having so much trouble with AP-STA, get a second wireless adapter (they're cheap) and use this instead. Good luck.