RaspAP/raspap-webgui #1108: web not showing Hourly traffic amount

progamer562, 2022-01-22:
when i login the first thing it shows me Hourly traffic amount i have been connecting to it but its not showing how much i useing it show this 
anyway to fix 

let me know if you need any debugging or something 

[image]

goldmojo, 2022-01-22:
Same here, I just updated from 2.6.6 to 2.8.3 and lost all traffic information.

goldmojo, 2022-01-22:
After a clean uninstall using the uninstallation script + fresh installation, I still have no traffic information and logs are limited to : 

2022-01-22 16:25:56: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: sidebarToggled in /var/www/html/includes/functions.php on line 683
2022-01-22 16:25:56: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined variable: arrHostapdConf in /var/www/html/templates/dashboard.php on line 96
2022-01-22 16:25:56: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: hour in /var/www/html/ajax/bandwidth/get_bandwidth_hourly.php on line 43
2022-01-22 16:25:56: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  count(): Parameter must be an array or an object that implements Countable in /var/www/html/ajax/bandwidth/get_bandwidth_hourly.php on line 44

closed 2022-01-22 by billz