RaspAP/raspap-webgui #1109: OpenVPN won't connect Unable to save OpenVPN client config

mjdcurtis, 2022-01-22:
## Checklist

- [x] This is a bug report
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

OpenVPN won't connect

Using ProtonVPN with the Username and password authentification method, OpenVPN won't start.

Unable to save client auth credentials

Enabling auth-user-pass in OpenVPN client.conf

Updating auth-user-passChecking iptables rules

Rule already exits: -A POSTROUTING -o tun0 -j MASQUERADE

Rule already exits: -A FORWARD -i tun0 -o wlan0 -m state --state 

RELATED,ESTABLISHED -j ACCEPT

Rule already exits: -A FORWARD -i wlan0 -o tun0 -j 

ACCEPT

Unable to save OpenVPN client config

## Your environment
1. Operating System: **ENTER HERE**  Raspbian GNU/Linux 10 (buster)
2. Hardware and version:  PiZero W
3. RaspAP version:  RaspAP v2.8.3
4. Followed the project prerequisites?  Yes
5. RaspAP Quick Install or Manual setup?  Quick Install
6. Using default configuration?  Yes
7. Simultaneous AP and managed mode?  No
8. Onboard wireless chipset or external adapter?  onboard
9. Other software or services running with RaspAP? 

## Steps to reproduce

Using ProtonVPN with the Username and password authentification method,

Unlike Issue #641, openvpn did install.

## Screenshots

## Additional context

systemctl status openvpn@server.service
● openvpn@server.service - OpenVPN connection to server
   Loaded: loaded (/etc/systemd/system/openvpn@.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2022-01-22 10:15:49 GMT; 2min 51s ago
     Docs: man:openvpn(8)
           https://community.openvpn.net/openvpn/wiki/Openvpn24ManPage
           https://community.openvpn.net/openvpn/wiki/HOWTO
 Main PID: 494 (openvpn)
   Status: "Initialization Sequence Completed"
    Tasks: 1 (limit: 877)
   CGroup: /system.slice/system-openvpn.slice/openvpn@server.service
           └─494 /usr/sbin/openvpn --daemon ovpn-server --status /run/openvpn/server.status 10 --cd /etc/openvpn --config /etc/openvpn/server.conf --writepid /run/openvpn/server.pid

 journalctl -xe
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- An ExecStart= process belonging to unit openvpn@client.service has exited.
[... 103 lines cut ...]
Jan 22 10:17:21 raspberrypi sudo[830]: www-data : TTY=unknown ; PWD=/var/www/html/ajax/openvpn ; USER=root ; COMMAND=/bin/rm /etc/openvpn/client/login.conf
Jan 22 10:17:21 raspberrypi systemd[1]: openvpn@client.service: Main process exited, code=exited, status=1/FAILURE
Jan 22 10:17:21 raspberrypi systemd[1]: openvpn@client.service: Failed with result 'exit-code'.
Jan 22 10:17:21 raspberrypi lighttpd[564]: rm: cannot remove '/etc/openvpn/client/login.conf': No such file or directory
Jan 22 10:17:21 raspberrypi sudo[835]: www-data : command not allowed ; TTY=unknown ; PWD=/var/www/html/ajax/openvpn ; USER=root ; COMMAND=/bin/ln -s /etc/openvpn/client/us.protonvpn.com.udp_login.conf /etc/openvpn/client/login.conf
Jan 22 10:17:22 raspberrypi sudo[837]: www-data : TTY=unknown ; PWD=/var/www/html/ajax/openvpn ; USER=root ; COMMAND=/bin/systemctl stop openvpn-client@client
Jan 22 10:17:22 raspberrypi sudo[840]: www-data : TTY=unknown ; PWD=/var/www/html/ajax/openvpn ; USER=root ; COMMAND=/bin/systemctl start openvpn-client@client
Jan 22 10:17:23 raspberrypi openvpn[842]: Options error: In [CMD-LINE]:1: Error opening configuration file: client.conf
Jan 22 10:17:23 raspberrypi openvpn[842]: Use --help for more information.
Jan 22 10:17:23 raspberrypi systemd[1]: openvpn-client@client.service: Main process exited, code=exited, status=1/FAILURE
Jan 22 10:17:23 raspberrypi systemd[1]: openvpn-client@client.service: Failed with result 'exit-code'.
Jan 22 10:17:23 raspberrypi sudo[844]: www-data : TTY=unknown ; PWD=/var/www/html/ajax/openvpn ; USER=root ; COMMAND=/bin/systemctl enable openvpn-client@client
Jan 22 10:17:23 raspberrypi lighttpd[564]: Job for openvpn-client@client.service failed because the control process exited with error code.
Jan 22 10:17:23 raspberrypi lighttpd[564]: See "systemctl status openvpn-client@client.service" and "journalctl -xe" for details.
Jan 22 10:17:28 raspberrypi sudo[891]: www-data : TTY=unknown ; PWD=/var/www/html ; USER=root ; COMMAND=/bin/mv /tmp/ovpn/authdata /etc/openvpn/client/us.protonvpn.com.udp_login.conf
Jan 22 10:17:28 raspberrypi sudo[894]: www-data : TTY=unknown ; PWD=/var/www/html ; USER=root ; COMMAND=/bin/rm /etc/openvpn/client/login.conf
Jan 22 10:17:28 raspberrypi lighttpd[564]: rm: cannot remove '/etc/openvpn/client/login.conf': No such file or directory
Jan 22 10:17:29 raspberrypi sudo[897]: www-data : command not allowed ; TTY=unknown ; PWD=/var/www/html ; USER=root ; COMMAND=/bin/ln -s /etc/openvpn/client/us.protonvpn.com.udp_login.conf /etc/openvpn/client/login.conf
Jan 22 10:17:29 raspberrypi sudo[899]: www-data : TTY=unknown ; PWD=/var/www/html ; USER=root ; COMMAND=/etc/raspap/openvpn/configauth.sh /tmp/ovpn/2240e81372bb9190507e9f971816dfc1a616d5321642846648 1 wlan0
Jan 22 10:17:30 raspberrypi sudo[908]: www-data : TTY=unknown ; PWD=/var/www/html ; USER=root ; COMMAND=/bin/mv /tmp/ovpn/2240e81372bb9190507e9f971816dfc1a616d5321642846648 /etc/openvpn/client/us.protonvpn.com.udp_client.conf
Jan 22 10:17:30 raspberrypi sudo[911]: www-data : TTY=unknown ; PWD=/var/www/html ; USER=root ; COMMAND=/bin/rm /etc/openvpn/client/client.conf
Jan 22 10:17:30 raspberrypi lighttpd[564]: rm: cannot remove '/etc/openvpn/client/client.conf': No such file or directory
Jan 22 10:17:31 raspberrypi sudo[914]: www-data : command not allowed ; TTY=unknown ; PWD=/var/www/html ; USER=root ; COMMAND=/bin/ln -s /etc/openvpn/client/us.protonvpn.com.udp_client.conf /etc/openvpn/client/client.conf
Jan 22 10:17:33 raspberrypi sudo[925]: www-data : TTY=unknown ; PWD=/var/www/html ; USER=root ; COMMAND=/etc/raspap/openvpn/openvpnlog.sh
Jan 22 10:17:33 raspberrypi sudo[930]:     root : TTY=unknown ; PWD=/var/www/html ; USER=root ; COMMAND=/usr/bin/tee /tmp/openvpn.log

mjdcurtis, 2022-01-24:
Deleted /etc/openvpn/client. Uninstalled everything related to openvpn I could find. Ran quick installer script for RaspAP, same problem.

```
Unable to save client auth credentials
Enabling auth-user-pass in OpenVPN client.conf
Updating auth-user-passChecking iptables rules
Rule already exits: -A POSTROUTING -o tun0 -j MASQUERADE
Rule already exits: -A FORWARD -i tun0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
Rule already exits: -A FORWARD -i wlan0 -o tun0 -j ACCEPTUnable to save OpenVPN client config
```

Log:
```

Jan 24 19:03:02 raspberrypi ovpn-client[600]: Options error: In [CMD-LINE]:1: Error opening configuration file: /etc/openvpn/client.conf
Jan 24 19:03:02 raspberrypi openvpn[602]: Options error: In [CMD-LINE]:1: Error opening configuration file: client\x03.conf
Jan 24 19:03:02 raspberrypi openvpn[602]: Use --help for more information.
Jan 24 19:03:02 raspberrypi openvpn[608]: Options error: In [CMD-LINE]:1: Error opening configuration file: server.conf
Jan 24 19:03:02 raspberrypi openvpn[608]: Use --help for more information.
Jan 24 19:03:03 raspberrypi systemd[1]: openvpn@client.service: Main process exited, code=exited, status=1/FAILURE
Jan 24 19:03:03 raspberrypi systemd[1]: openvpn@client.service: Failed with result 'exit-code'.
Jan 24 19:03:03 raspberrypi openvpn[621]: Options error: In [CMD-LINE]:1: Error opening configuration file: client.conf
Jan 24 19:03:03 raspberrypi openvpn[621]: Use --help for more information.
Jan 24 19:03:03 raspberrypi systemd[1]: openvpn-client@client\x03.service: Main process exited, code=exited, status=1/FAILURE
Jan 24 19:03:03 raspberrypi systemd[1]: openvpn-client@client\x03.service: Failed with result 'exit-code'.
Jan 24 19:03:04 raspberrypi systemd[1]: openvpn-server@server.service: Main process exited, code=exited, status=1/FAILURE
Jan 24 19:03:04 raspberrypi systemd[1]: openvpn-server@server.service: Failed with result 'exit-code'.
Jan 24 19:03:04 raspberrypi systemd[1]: openvpn-client@client.service: Main process exited, code=exited, status=1/FAILURE
Jan 24 19:03:04 raspberrypi systemd[1]: openvpn-client@client.service: Failed with result 'exit-code'.
Jan 24 19:03:08 raspberrypi systemd[1]: openvpn@client.service: Service RestartSec=5s expired, scheduling restart.
Jan 24 19:03:08 raspberrypi systemd[1]: openvpn@client.service: Scheduled restart job, restart counter is at 1.
Jan 24 19:03:09 raspberrypi systemd[1]: openvpn-server@server.service: Service RestartSec=5s expired, scheduling restart.
Jan 24 19:03:09 raspberrypi systemd[1]: openvpn-server@server.service: Scheduled restart job, restart counter is at 1.
Jan 24 19:03:24 raspberrypi ovpn-client[715]: Options error: In [CMD-LINE]:1: Error opening configuration file: /etc/openvpn/client.conf
Jan 24 19:03:24 raspberrypi systemd[1]: openvpn@client.service: Main process exited, code=exited, status=1/FAILURE
Jan 24 19:03:24 raspberrypi systemd[1]: openvpn@client.service: Failed with result 'exit-code'.
Jan 24 19:03:24 raspberrypi openvpn[725]: Options error: In [CMD-LINE]:1: Error opening configuration file: server.conf
Jan 24 19:03:24 raspberrypi openvpn[725]: Use --help for more information.
Jan 24 19:03:24 raspberrypi systemd[1]: openvpn-server@server.service: Main process exited, code=exited, status=1/FAILURE
Jan 24 19:03:24 raspberrypi systemd[1]: openvpn-server@server.service: Failed with result 'exit-code'.
Jan 24 19:03:29 raspberrypi systemd[1]: openvpn@client.service: Service RestartSec=5s expired, scheduling restart.
Jan 24 19:03:29 raspberrypi systemd[1]: openvpn@client.service: Scheduled restart job, restart counter is at 2.
Jan 24 19:03:30 raspberrypi systemd[1]: openvpn-server@server.service: Service RestartSec=5s expired, scheduling restart.
Jan 24 19:03:30 raspberrypi systemd[1]: openvpn-server@server.service: Scheduled restart job, restart counter is at 2.
Jan 24 19:03:30 raspberrypi ovpn-client[791]: Options error: In [CMD-LINE]:1: Error opening configuration file: /etc/openvpn/client.conf
Jan 24 19:03:30 raspberrypi systemd[1]: openvpn@client.service: Main process exited, code=exited, status=1/FAILURE
Jan 24 19:03:30 raspberrypi systemd[1]: openvpn@client.service: Failed with result 'exit-code'.
Jan 24 19:03:31 raspberrypi openvpn[792]: Options error: In [CMD-LINE]:1: Error opening configuration file: server.conf
Jan 24 19:03:31 raspberrypi openvpn[792]: Use --help for more information.
Jan 24 19:03:31 raspberrypi systemd[1]: openvpn-server@server.service: Main process exited, code=exited, status=1/FAILURE
Jan 24 19:03:31 raspberrypi systemd[1]: openvpn-server@server.service: Failed with result 'exit-code'.
Jan 24 19:03:35 raspberrypi systemd[1]: openvpn@client.service: Service RestartSec=5s expired, scheduling restart.
Jan 24 19:03:35 raspberrypi systemd[1]: openvpn@client.service: Scheduled restart job, restart counter is at 3.
Jan 24 19:03:36 raspberrypi ovpn-client[847]: Options error: In [CMD-LINE]:1: Error opening configuration file: /etc/openvpn/client.conf
Jan 24 19:03:36 raspberrypi systemd[1]: openvpn-server@server.service: Service RestartSec=5s expired, scheduling restart.
Jan 24 19:03:36 raspberrypi systemd[1]: openvpn-server@server.service: Scheduled restart job, restart counter is at 3.
Jan 24 19:03:36 raspberrypi systemd[1]: openvpn@client.service: Main process exited, code=exited, status=1/FAILURE
Jan 24 19:03:36 raspberrypi systemd[1]: openvpn@client.service: Failed with result 'exit-code'.
[... cut ...]
Jan 24 19:05:00 raspberrypi systemd[1]: openvpn@client.service: Main process exited, code=exited, status=1/FAILURE
Jan 24 19:05:00 raspberrypi systemd[1]: openvpn@client.service: Failed with result 'exit-code'.
Jan 24 19:05:00 raspberrypi openvpn[1091]: Options error: In [CMD-LINE]:1: Error opening configuration file: server.conf
Jan 24 19:05:00 raspberrypi openvpn[1091]: Use --help for more information.

```

I will try a manual install of openvpn before using the RaspAP quickinstall script I suppose?

billz, 2022-01-24:
I'd rather suggest a fresh install per the project prerequisites. The presence of openvpn-server and server.conf indicates this is NOT a clean install. 

```
Options error: In [CMD-LINE]:1: Error opening configuration file: server.conf
```

Following that, as stated: 

> follow your provider's manual instructions for Linux. If you get the same error, resolve it first before using RaspAP.

RaspAP provides a front-end to this service only. If it fails to work from the shell, do not expect different results from the UI.

mjdcurtis, 2022-01-24:
I will try that thanks.

mjdcurtis, 2022-01-26:
Clean install worked. Not sure what went wrong the first time.

closed 2022-01-26 by mjdcurtis

billz, 2022-01-26:
Glad to hear it.

It's for this reason that we ask users to follow the [project prerequisites](https://docs.raspap.com/#quick-start) before opening a new issue.

mjdcurtis, 2022-01-26:
I thought the install was clean but I was clearly mistaken. Thanks, I really appreciate the help!