RaspAP/raspap-webgui #1119: Failed to open stream /etc/raspap/hostapd.ini

Tajnymag, 2022-02-09:
## Checklist

- [ ] This is a bug report
- [ ] I have searched this repository for existing issues
- [ ] I checked the FAQ and official documentation before creating this issue
- [ ] I have read and understand the issue reporting guidelines 

## Bug description

## Your environment
1. Operating System: RPi OS Lite 11/Bullseye
2. Hardware and version: RPi 3B+
3. RaspAP version: 2.8.4
4. Followed the project prerequisites? Yes
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? Yes, without OpenVPN and Wireguard
7. Simultaneous AP and managed mode? No
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP? No

## Steps to reproduce

a) install a fresh image of RPi OS Lite 11
b) upgrade the system (full-upgrade as told by README)
c) run quick installer and deselect OpenVPN/Wireguard
d) reboot and connect to raspi-webgui wlan
e) open http://10.3.141.1 in browser and login with credentials from README
f) check how the page is rendered and that the file also doesn't exist on the filesystem

## Screenshots

[image]

## Additional context

OK, my context isn't completely crystal clean.
Firstly: I am connected to a wired 802.1x network by using a configuration and hook from [this](https://forums.raspberrypi.com/viewtopic.php?t=285379) forum post.
Secondly: During the first quick install run, the Pi encountered a kernel panic and I had to forcefully restart it. The second time I ran the installer, the installation went fine. After encountering the errors in the web config, I've also tried to uninstall raspap and install it again. Didn't solve the issue.

billz, 2022-02-09:
Your bug report is missing some vital information, including the full error (assuming `Failed to open stream: Permission denied`). 

Your install is incomplete and/or you have other issues with your device. Start with a fresh SD and follow the project prerequisites.

closed 2022-02-09 by billz

Tajnymag, 2022-02-09:
I have followed the prerequisites.
I have started with as fresh installation as possible to be even able to connect to the internet.

Where are the error logs located?

billz, 2022-02-09:
https://docs.raspap.com/ap-basics/#troubleshooting

This is not a troubleshooting forum. https://docs.raspap.com/issues/