RaspAP/raspap-webgui #1150: Cannot get wlan0 to route to wlan1 (USB WIFI), eth0 works

rhukster, 2022-03-31:
## Checklist

- [x] This is a bug report
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

## Your environment
1. Operating System:  RPi OS 32-bit Lite
2. Hardware and version: RPi 4b
3. RaspAP version: v2.8.3
4. Followed the project prerequisites? Yes
5. RaspAP Quick Install or Manual setup? Quick
6. Using default configuration? Yes
7. Simultaneous AP and managed mode? No
8. Onboard wireless chipset or external adapter? Both
9. Other software or services running with RaspAP? 

## Steps to reproduce

I've been struggling to get RaspAP working with `wlan0` using built-in WIFI and service as the hotspot/dhcp interface but with a USB WIFI adapter for `wlan1` as the client.  When I install a wired connection via `eth0` things work as expected.  So the built-in Hotspot with DHCP is working as expected (using 802.11ac).

I've tried 2 different USB wifi adapters, a Netgear A6210 which was detected and configured out of the box (not sure the driver it used but it was able to connect to my local WIFI network no problem).  I also have just tried a Cudy WU1300S that required me to use the `install_wlan_drivers.sh` script to build the `8rtl88x2bu` drivers.  Again after doing that and rebooting it picked up my  local area WIFI no probem.

The issue is that when i disconnect my `eth0` port (rebooting makes no difference), traffic from the hotspot clients is not able to route through the `wlan1` adapter.  Plugging the cable back in resolves the issue.

I've tried searching for this setup but have not really found anything that explains how the internal routing should be configured to support a WIFI client as well as an ethernet based client.  The dashboard shows only info on the WIFI client, and that all looks correct.

## Screenshots

Here are some screenshots when the `eth0` is plugged in (easiest for me to take screenshots with):

[image]

[image]

[image]

## Additional context

I can provide any additional info needed, just let me know what commands to run!

Thanks so much!

avierywan, 2022-04-01:
Exact same issue. When I tried swapping interface wlan0 and wlan1, wlan1 appears to be unable to get internet access; wlan0 (built in wifi) is able to reach the internet. The pi can still be reached through wlan1 using SSH, but other functions such as VNC do not work. After installing raspAP, wlan1 will show up as disabled due to "wifi country not set" in the pi desktop.

Operating System: Raspberry Pi OS 64-bit
Dongle chip: RTL8821AU
Everything else same as OP

rhukster, 2022-04-01:
It definitely seems to be related to 802.11ac 5ghz USB adapter because I tried an old 802.11n adapter and it works just fine.

billz, 2022-04-01:
Relevant FAQ https://docs.raspap.com/faq/#interfaces

billz, 2022-04-01:
Please be sure you've followed the steps outlined above.
@avierywan also note that distros other than those listed by this project are completely unsupported.

billz, 2022-04-01:
Duplicate of https://github.com/RaspAP/raspap-webgui/issues/335. See also this discussion https://github.com/RaspAP/raspap-webgui/discussions/1022. 

Please check for existing issues before reporting, thanks

closed 2022-04-01 by billz