RaspAP/raspap-webgui #1161: Internet traffic not being routed to OpenVPN tunnel interface

efierro, 2022-04-24:
Followed the quick install instructions in a Raspberri Pi 4 and enabled OpenVPN with username/password and an .ovpn file.
All looks good, no errors I can see, the RaspAPI portal shows OpenVPN up and running but nothing is being routed through the tunnel.
Using iftop -i tun0 does not show any traffic.
I have two wireless interfaces, built-in wlan0 and an external USB2 Ralink RT5370 wlan1.
I think the issue with routing is related to this:
```
root@rsbpi:~# iptables -A POSTROUTING -o tun0 -j MASQUERADE
iptables: No chain/target/match by that name.
```
These are my current iptables settings:
```
root@rsbpi:~# iptables -L
Chain INPUT (policy ACCEPT)
target     prot opt source               destination

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination
ACCEPT     all  --  anywhere             anywhere             state RELATED,ESTABLISHED
ACCEPT     all  --  anywhere             anywhere

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
root@rsbpi:~# iptables -L -t nat
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination

Chain INPUT (policy ACCEPT)
target     prot opt source               destination

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination
MASQUERADE  all  --  anywhere             anywhere
MASQUERADE  all  --  192.168.50.0/24     !192.168.50.0/24
MASQUERADE  all  --  anywhere             anywhere
MASQUERADE  all  --  anywhere             anywhere
root@rsbpi:~#
```

Setup:

1. Operating System: RPi OS 32-bit Lite
2. Hardware and version: RPi 4 B 8G
3. RaspAP version: 2.8.4
4. Followed the project prerequisites? Yes
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? Yes
7. Simultaneous AP and managed mode? No
8. Onboard wireless chipset or external adapter? both, see above
9. Other software or services running with RaspAP? No

billz, 2022-04-25:
OpenVPN was tested and validated with this release.

The following is displayed in the UI on saving an OpenVPN config:

```
Adding rule: -t nat -A POSTROUTING -o tun0 -j MASQUERADE
Adding rule: -A FORWARD -i tun0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
Adding rule: -A FORWARD -i wlan0 -o tun0 -j ACCEPT
```

These rules may be listed like so:

```
$ sudo iptables -L FORWARD -v -n
Chain FORWARD (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         
 1955 1493K ACCEPT     all  --  tun0   wlan0   0.0.0.0/0            0.0.0.0/0            state RELATED,ESTABLISHED
 1715  194K ACCEPT     all  --  wlan0  tun0    0.0.0.0/0            0.0.0.0/0    
```

See also https://docs.raspap.com/openvpn/#troubleshooting

efierro, 2022-04-25:
Output ok when saving the OpenVPN config.
Rules listed for FORWARD also good.
Tried all troubleshooting points in the documentation.
Still did not work. Giving up.
Thanks