RaspAP/raspap-webgui #1164: Adding a new WiFi client always gives WPA passphrase must be between 8 and 63 characters

madv-zz, 2022-05-02:
## Checklist

- [ x] This is a bug report
- [ x] I have searched this repository for existing issues
- [ x] I checked the FAQ and official documentation before creating this issue
- [ x] I have read and understand the issue reporting guidelines 

## Bug description

## Your environment
1. Operating System:  RPi OS 32-bit Lite 
2. Hardware and version: RPi4 
3. RaspAP version: 2.8.4 
4. Followed the project prerequisites? Yes 
5. RaspAP Quick Install or Manual setup? Install
6. Using default configuration? Yes 
7. Simultaneous AP and managed mode? No 
8. Onboard wireless chipset or external adapter? Both 
9. Other software or services running with RaspAP? No / drivers for 

## Steps to reproduce

install Raspbian
install RaspAP and thus full upgrade

Try to add a client network you will get the message that your passphrase isn't between 8 and 63 char. Even when it is.

## Screenshots

## Additional context

For now editing /etc/wpa_supplicant/wpa_supplicant.conf by hand to add a client network.

ls -slah /etc/wpa_supplicant/wpa_supplicant.conf
4.0K -rw------- 1 root root 396 May  2 10:26 /etc/wpa_supplicant/wpa_supplicant.conf

the error messages comes from /var/www/html/includes/configure_client.php

billz, 2022-05-02:
Please provide detail on your steps to repo. 

https://github.com/RaspAP/raspap-webgui/blob/993e65483d9820ae666c5086e5ffb4d4860c4422/includes/configure_client.php#L61

madv-zz, 2022-05-02:
Repo staps are:
Install the current 32 bit Raspbian using the pi imager.
Use the quick start to install RaspAP
Add an USB wifi adapter so you have a dedicate client adapter.
Add a client network.

You won't be able to add the client network as it gives the error about a to short passphrase.

billz, 2022-05-02:
No issues adding a client with psk = 12345678 or a random string of 63 chars.
Your psk above is 65 characters. Attempting to add this will return the message you're seeing.

Closing this. Refer to https://docs.raspap.com/faq/#what-are-the-passphrase-requirements-used-by-raspap
Thanks

closed 2022-05-02 by billz

madv-zz, 2022-05-02:
Example:

wpa_passphrase 1 12345678
network={
	ssid="1"
	#psk="12345678"
	psk=2f44de4e9197f92dead8b3560e8a0378c43e13477fa1154b5c6ff6b4f17889b3
}

So wpa_passphrase generates a faulty encoded PSK?

The example above is made using
 raspap-webgui/includes/configure_client.php
 The part i guess = exec('wpa_passphrase '. ssid2utf8( escapeshellarg($ssid) ) . ' ' . escapeshellarg($network['passphrase']), $wpa_passphrase);