RaspAP/raspap-webgui #1168: 404 on all pages except index

johncarterofmars, 2022-05-07:
## Checklist

- [x ] This is a bug report
- [x ] I have searched this repository for existing issues
- [x ] I checked the FAQ and official documentation before creating this issue
- [x ] I have read and understand the issue reporting guidelines 

## Bug description

I updated to 2.8.4 today using: 
curl -sL https://install.raspap.com | bash -s -- --upgrade
on a functioning RP4. After the update and reboot, all pages for the webgui are 404 except for the dashboard. I reviewed other issues here and found one from two years ago concerning the /etc/lighttpd/conf-enabled/50-raspap-router.conf file. However, mine was already configured exactly as the one in the solution. Here are some of the errors from lighttpd:

2022-05-07 14:52:08: mod_fastcgi.c.487) FastCGI-stderr:PHP Warning:  parse_ini_file(/etc/raspap/hostapd.ini): failed to open stream: No such file or directory in /var/www/html/includes/wifi_functions.php on line 160
2022-05-07 14:52:08: mod_fastcgi.c.487) FastCGI-stderr:PHP Warning:  parse_ini_file(/etc/raspap/hostapd.ini): failed to open stream: No such file or directory in /var/www/html/includes/wifi_functions.php on line 160
2022-05-07 14:52:08: mod_fastcgi.c.487) FastCGI-stderr:PHP Notice:  Undefined index: PATH_INFO in /var/www/html/index.php on line 53
2022-05-07 14:52:08: mod_fastcgi.c.487) FastCGI-stderr:PHP Notice:  Undefined index: sidebarToggled in /var/www/html/includes/functions.php on line 683
2022-05-07 14:52:08: mod_fastcgi.c.487) FastCGI-stderr:PHP Warning:  parse_ini_file(/etc/raspap/hostapd.ini): failed to open stream: No such file or directory in /var/www/html/includes/functions.php on line 691
2022-05-07 14:52:08: mod_fastcgi.c.487) FastCGI-stderr:PHP Notice:  Trying to access array offset on value of type bool in /var/www/html/includes/functions.php on line 693
2022-05-07 14:52:08: mod_fastcgi.c.487) FastCGI-stderr:PHP Notice:  Undefined index: theme in /var/www/html/includes/functions.php on line 733
2022-05-07 14:52:08: mod_fastcgi.c.487) FastCGI-stderr:PHP Warning:  parse_ini_file(/etc/raspap/hostapd.ini): failed to open stream: No such file or directory in /var/www/html/includes/wifi_functions.php on line 160

## Your environment
1. Operating System: RPi4 8GB Raspberry Pi OS 64bit * 
2. Hardware and version: RPI 4 8GB 
3. RaspAP version: 2.8.4 
4. Followed the project prerequisites? Yes 
5. RaspAP Quick Install or Manual setup? Upgrade 
6. Using default configuration? yes 
7. Simultaneous AP and managed mode? 
8. Onboard wireless chipset or external adapter? both 
9. Other software or services running with RaspAP? zabbix agent, Zero Tier one

## Steps to reproduce

Rand the update script on a working device and this is the result
## Screenshots

## Additional context

billz, 2022-05-08:
The upgrade does not touch your existing lighttpd config. If your web server and/or contents of `/var/www/html` have been modified (perhaps by other installed software?) you will need to apply RaspAP's routing rules as described here:
https://docs.raspap.com/manual/#create-the-web-application

Alternatively, do a fresh install of RaspAP and specify a custom path when prompted.

Note that Zerotier has not been tested with this project and may be incompatible.

johncarterofmars, 2022-05-14:
Thank you. I ended up having to purge lighttpd and removing /var/www and then reinstalling from scratch. It worked then. I appreciate your help.

closed 2022-05-30 by billz