RaspAP/raspap-webgui #1169: Additional things to get it running on a stock Pi4

brettbeeson, 2022-05-11:
## Checklist

- Not really a bug report, but not a feature request!
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

## Your environment
1. Operating System: RPi OS 32-bit (stock via Raspi Imager) 
2. Hardware and version: RPi 4
3. RaspAP version: RaspAP v2.8.4
4. Followed the project prerequisites? Yes
5. RaspAP Quick Install or Manual setup? Quick
6. Using default configuration? No - selected "Yes" to everything except Ad Block and OpenVPN
7. Simultaneous AP and managed mode? No
8. Onboard wireless chipset or external adapter? Both
9. Other software or services running with RaspAP? No

## Steps to reproduce
1. Stock install of Raspian on a Pi 4.
2. Add Wifi dongle, hence wlan0 = on-board and wlan1 = dongle
3. Run the quick installer and reboot

## Result (two issues)

Clients can see raspi-gui and connect, but no IP is given. I see that systemd-resolve is listening on port 53 and dnsmasq cannot start. I add `DNSStubLIstener=no # /etc/systemd/resolve.conf` and restart. dnsmasq runs okay now.

Secondly, there is no internet available because first route is 10.3.141.1. Edit of `/etc/dhcpcd.conf` and push wlan0 down a bit by adding `metric 500` under wlan1 (which is connected to internet).

Now it works.

billz, 2022-05-11:
These steps are not necessary on RPi OS with a clean install.

`systemd-networkd` is stopped and disabled by the installer. If `sudo systemctl status systemd-networkd` shows it's active, perhaps you, or another software package, has enabled it for some reason. This will not allow RaspAP to function in normal (routed) AP mode.

closed 2022-05-11 by billz

brettbeeson, 2022-05-12:
Hi Billz, thanks for the feedback. I didn't (re)enable it. I guess something else did.