RaspAP/raspap-webgui #1173: Install function references directories that haven't been created yet

CamDavidsonPilon, 2022-05-22:
## Checklist

- [x] This is a bug report
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description
I noticed the following in the install logs:
```
RaspAP Install: Backing up existing configs to /etc/raspap/backups
/usr/bin/cp: cannot create regular file '/etc/raspap/backups/interfaces.2022-05-22-23:14': No such file or directory
/usr/bin/ln: failed to create symbolic link '/etc/raspap/backups/interfaces': No such file or directory
/usr/bin/cp: cannot create regular file '/etc/raspap/backups/dhcpcd.conf.2022-05-22-23:14': No such file or directory
/usr/bin/ln: failed to create symbolic link '/etc/raspap/backups/dhcpcd.conf': No such file or directory
[ \u2713 ok ]
RaspAP Install: Creating RaspAP directories
Creating /etc/raspap/networking
Adding /etc/dhcpcd.conf as base configuration
Changing file ownership of /etc/raspap
```

`_check_for_old_configs` comes first in the [setup list](https://github.com/RaspAP/raspap-webgui/blob/6e3177ea318128a9026cb5ec2ae728a6b72b8356/installers/common.sh#L46-L47), but this [references](https://github.com/RaspAP/raspap-webgui/blob/6e3177ea318128a9026cb5ec2ae728a6b72b8356/installers/common.sh#L457-L489) directories that haven't been created yet. 

## Your environment
1. QEMU
3. version 2.8.4
4. No?
5. Quick Install
6. Yes?
7. No
8. Onboard

billz, 2022-05-22:
This is a known issue https://github.com/RaspAP/raspap-webgui/pull/1162

CamDavidsonPilon, 2022-05-22:
So it is! Okay, closing this.

closed 2022-05-22 by CamDavidsonPilon