RaspAP/raspap-webgui #1174: Support for Zerotier (similar to WireGuard and OpenVPN)

AleXSR700, 2022-05-23:
**Is your feature request related to a problem? Please describe.**
Zerotier installed individually and RaspAP installed individually are not working together. Similar to VPN solutions, the need for proper bridging etc. can result in the networks not working properly.
Out-of-the-box support by RaspAP would prevent these issues.

**Describe the solution you'd like**
Integration of Zerotier into RaspAP (similar to OpenVPN and WireGuard).

**Describe alternatives you've considered**
Tried a dozen approaches to get them working individually. None worked. Only implementation into RaspAP looks like possible solution (since RaspAP devs are far far more experienced in networking under Linux than I am).

**Additional context**
Zerotier has become a very popular VPN alternative for all the users who, like me, do not get IPv4 adresses from the ISP (DS-Lite etc.). OpenVPN and other VPN solutions do not work e.g. in Germany when using Vodafone Cable as ISP. Zerotier is needed.
Without integration into RaspAP, users are forced to use either or :(

billz, 2022-05-23:
See my comment in your open discussion https://github.com/RaspAP/raspap-webgui/discussions/1166

AleXSR700, 2022-05-23:
Hello @billz ,
thank you, yes, I saw that comment. But I found some users who seem to have it working without issues. So I am not sure what is causing the issues in my case, but it seems possible to get it running.
There is even a [repository](https://github.com/rogeriopradoj/tuto-zerotier-raspap-in-my-rpi3b/blob/main/README.md), but I tried it and it did not work for me. So I am guessing that at some point the configs are messing with each other. 
But it seems possible. And it would be more likely to work if it were integrated into RaspAP because RaspAP would configure it accordingly.

I know it is a long shot, but a feature request is a long shot :)

billz, 2022-05-28:
No one in this project, myself included, has used this software.

Maybe instead of a "this doesn't work, make it work for me" support type request, you could try contacting the user whose repo you linked here. His repo was created last month, so I assume he's using bullseye, and his email is on his GitHub profile.

Protip: you will (usually) have better luck getting help if you can provide things like logs, process output or your kernel's routing table (use `route -n`) to assist with troubleshooting. Probably the Zerotier docs have some pointers on this.

AleXSR700, 2022-05-29:
Hi,
yes, I supplied all of that info on both reddit and on the zerotier forum. We were unable to solve it.

I will contact the owner of the github repo again but I had opened an issue about it with no luck.

jsn0327, 2022-12-23:
I have found a fix for this.  See my post [Here](https://github.com/RaspAP/raspap-webgui/issues/1270)

closed 2023-12-12 by github-actions[bot]