RaspAP/raspap-webgui #1193: webgui 403 forbidden but permissions look good and no hint in error log

riker65, 2022-07-26:
## Checklist

- [ x] This is a bug report
- [ ] I observed this bug on a clean install of the OS
- [ ] I have followed the project prerequisites
- [x ] I have searched this repository for existing issues
- [x ] I checked the FAQ and official documentation before creating this issue
- [x ] I have read and understand the issue reporting guidelines 

## Bug description

localhost website in chromium : error 403 forbidden

## Your environment
1. Operating System: **ENTER HERE** rpi3 debian 11
2. Hardware and version:  rpi3
3. RaspAP version:  latest
4. Clean install of a compatible operating system?  used clean install but it is corrupt somehow
5. RaspAP Quick Install or Manual setup?  install, check for error and did some manual steps 
6. Using default configuration?  yes
7. Simultaneous AP and managed mode?  no
8. Onboard wireless chipset or external adapter?   onboard and external
9. Other software or services running with RaspAP?  
10. 

## Steps to reproduce

launched the website localhost several times, rebooted

## Screenshots

## Additional context

permissions
`
drwxr-xr-x 13 root     root     4096 Jul 25 09:20 ..
drwxr-xr-x 13 www-data www-data 4096 Jul 26 17:00 html
drwxr-xr-x  2 root     root     4096 Jul 25 09:20 html.2022-07-25-09:20
pi@pi2Solar:/var/www$ cd html
pi@pi2Solar:/var/www/html$ ls -al
total 364
drwxr-xr-x 13 www-data www-data   4096 Jul 26 17:00 .
drwxr-xr-x  4 root     root       4096 Jul 26 16:55 ..
drwxr-xr-x  6 www-data www-data   4096 Jul 26 17:00 ajax
drwxr-xr-x  8 www-data www-data   4096 Jul 26 17:00 app
-rw-r--r--  1 www-data www-data   4365 Jul 26 17:00 BACKERS.md
-rw-r--r--  1 www-data www-data    946 Jul 26 17:00 composer.json
drwxr-xr-x  3 www-data www-data   4096 Jul 26 17:00 config
-rw-r--r--  1 www-data www-data     27 Jul 26 17:00 _config.yml
-rw-r--r--  1 www-data www-data   1081 Jul 26 17:00 CONTRIBUTING.md
-rw-r--r--  1 www-data www-data    140 Jul 26 17:00 crowdin.yml
drwxr-xr-x 11 www-data www-data   4096 Jul 26 17:00 dist
-rw-r--r--  1 www-data www-data   1150 Jul 26 17:00 favicon.ico
drwxr-xr-x  8 www-data www-data   4096 Jul 26 17:00 .git
drwxr-xr-x  4 www-data www-data   4096 Jul 26 17:00 .github
-rw-r--r--  1 www-data www-data     75 Jul 26 17:00 .gitignore
-rw-r--r--  1 www-data www-data   4279 Jul 26 17:00 gulpfile.js
drwxr-xr-x  2 www-data www-data   4096 Jul 26 17:00 includes
-rwxr-xr-x  1 www-data www-data  14468 Jul 26 17:00 index.php
drwxr-xr-x  2 www-data www-data   4096 Jul 26 17:00 installers
drwxr-xr-x  2 www-data www-data   4096 Jul 26 17:00 _layouts
-rw-r--r--  1 www-data www-data  35146 Jul 26 17:00 LICENSE
drwxr-xr-x 24 www-data www-data   4096 Jul 26 17:00 locale
-rw-r--r--  1 www-data www-data   1505 Jul 26 17:00 package.json
-rwxr-xr-x  1 www-data www-data    508 Jul 26 17:00 raspap.php
-rw-r--r--  1 www-data www-data  12958 Jul 26 17:00 README.md
drwxr-xr-x 10 www-data www-data   4096 Jul 26 17:00 templates
-rw-r--r--  1 www-data www-data    344 Jul 26 17:00 .travis.yml
-rw-r--r--  1 www-data www-data 194837 Jul 26 17:00 yarn.lock
`
error log

`022-07-26 10:18:15: server.c.1513) server started (lighttpd/1.4.59)
2022-07-26 10:19:00: server.c.1976) server stopped by UID = 0 PID = 1
2022-07-26 10:20:38: server.c.1513) server started (lighttpd/1.4.59)
2022-07-26 17:18:16: server.c.1976) server stopped by UID = 0 PID = 1
2022-07-26 17:18:18: server.c.1513) server started (lighttpd/1.4.59)
2022-07-26 17:45:52: server.c.1976) server stopped by UID = 0 PID = 1
2022-07-26 17:45:54: server.c.1513) server started (lighttpd/1.4.59)
2022-07-26 17:56:23: server.c.1976) server stopped by UID = 0 PID = 1
2022-07-26 17:57:14: server.c.1513) server started (lighttpd/1.4.59)`

riker65, 2022-07-26:
checked ports
`i@pi2Solar:~$ sudo netstat -tulpn | grep 80
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      674/lighttpd
tcp6       0      0 :::80                   :::*                    LISTEN      674/lighttpd
`

billz, 2022-07-26:
https://docs.raspap.com/issues/

closed 2022-07-26 by billz