RaspAP/raspap-webgui #1194: Problem with "your connection to this site is not Private" so i cant Login

LostiLama, 2022-07-27:
## Checklist

- [X] This is a bug report
- [X] I observed this bug on a clean install of the OS
- [X] I have followed the project prerequisites
- [X] I have searched this repository for existing issues
- [X] I checked the FAQ and official documentation before creating this issue
- [X] I have read and understand the issue reporting guidelines 

## Bug description

## Your environment
1. Operating System: **RPi OS 32-bit** 
2. Hardware and version: RPi 4 
3. RaspAP version:  
4. Clean install of a compatible operating system? Yes 
5. RaspAP Quick Install or Manual setup? Quick Install 
6. Using default configuration? Yes 
7. Simultaneous AP and managed mode? IDK 
8. Onboard wireless chipset or external adapter? Onboard 
9. Other software or services running with RaspAP? Yes no effect 

## Steps to reproduce

Problem with "your connection to this site is not Private" so i cant Login
## Screenshots

[image]

## Additional context

I cant login what can i do yesterday it worked. wifi is still up and it isnt the Browser so what can i do Fresh reinstall because on my second Pi there is the same Problem.

billz, 2022-07-27:
Your browser and/or plugins are raising this warning, not RaspAP. Try another browser or disable plugins that enforce HTTPS-only connections.

closed 2022-07-27 by billz

LostiLama, 2022-07-27:
I also tried to reinstall the Operating system and nothing changed what can i do it worked yesterday even tried to change the setting in the Installation nothing helped