RaspAP/raspap-webgui #1200: raspap pihole

PA3DSS, 2022-07-29:
## Checklist

- [ ] This is a bug report
- [ ] I observed this bug on a clean install of the OS
- [ ] I have followed the project prerequisites
- [ ] I have searched this repository for existing issues
- [ ] I checked the FAQ and official documentation before creating this issue
- [ ] I have read and understand the issue reporting guidelines 

## Bug description

## Your environment
1. Operating System: bullseye 
2. Hardware and version: RPi4 
3. RaspAP version: 2.8.5
4. Clean install of a compatible operating system? Yes 
5. RaspAP Quick Install or Manual setup? quick install
6. Using default configuration? yes
7. Simultaneous AP and managed mode? yes
8. Onboard wireless chipset or external adapter? onboard 
9. Other software or services running with RaspAP? no

## Steps to reproduce

webgui: DHCP server -->advanced
## Screenshots

[image]

## Additional context

raspap works correct with DNS server not pointing to my pihole but connected devices by my LAN, laptop or whatever connected via LAN (UTP cable) works perfect, but not my WiFi device, if i change raspap DNS to 1.1.1.1 my WiFi device works ok, i would like that all DNS query's going to pihole, by the way, i use the latest pihole version which is running on an other pi.

What is going wrong ? did i made a mistake ?

billz, 2022-07-29:
https://docs.raspap.com/faq/#pihole

PA3DSS, 2022-07-29:
thanks for the fast reply, now dnsmasq won't start anymore, rebooted with default settings, doesn't restart, i did exactly what's in your documentation, i think tomorrow all over again with a fresh install, see what happend

PA3DSS, 2022-07-29:
i can see all DNS query's in pihole now, but the screen keeps white wth this configuration
[image]

PA3DSS, 2022-07-29:
if i change DNS back to 1.1.1.1 it works perfect
[image]

version Pi-hole v5.11.4  FTL v5.16.1 Web Interface v5.13

closed 2022-07-30 by billz

PA3DSS, 2022-07-30:
With a fresh install of bullseye, reboot after running updates, during install raspap with this command
curl -sL https://install.raspap.com | bash
during install i get an error about hostapd, i stopped the install procedure with ctrl-s.

[image]
Continued  install procedure with ctrl-q during this is the second fresh install, i didn't notice during the first install and got the error in the web-gui that hostapd not started and i couldn't find whats going on, that's why i did a second fresh install.
[image]

What to do now ?

billz, 2022-07-30:
Please respect our issue policy. This is not a troubleshooting forum, thanks.