RaspAP/raspap-webgui #1201: Devices with static ip are not displayed

DerDoubleD, 2022-07-30:
## Checklist

- [ x] This is a bug report
- [x ] I observed this bug on a clean install of the OS
- [ x] I have followed the project prerequisites
- [ x] I have searched this repository for existing issues
- [x ] I checked the FAQ and official documentation before creating this issue
- [ x] I have read and understand the issue reporting guidelines 

## Bug description
Devices with a static ip are not displayed in the overview!

## Your environment
1. Operating System: RPi OS 32-bit Lite
2. Hardware and version: RPi 3B+
3. RaspAP version: RaspAP v2.8.5
4. Clean install of a compatible operating system? Yes
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration?  Yes
7. Simultaneous AP and managed mode? No ????
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP? No

## Steps to reproduce
I connected 4 devices with static IP to raspAP. The devices are not displayed but are connected.

## Screenshots
[image]
[image]

## Additional context
I connected 4 devices with static IP to raspAP. The devices are not displayed but are connected.

billz, 2022-07-30:
RaspAP uses the contents of `/var/lib/misc/dnsmasq.leases` and the output of `iw` to create the client list. It does not rely on the ARP cache.

https://github.com/RaspAP/raspap-webgui/blob/ee3de983ab37be73dd9044d9ccf771f0870ce44c/includes/dashboard.php#L193

iakiakiak, 2023-04-14:
A solution could be my feature request #1332  .

closed 2023-12-12 by github-actions[bot]