RaspAP/raspap-webgui #1205: no internet connection of wifi client AP hotspot on RetroPie OS

diegoxcn, 2022-08-17:
## Checklist

- [ ] This is a bug report  ---   I'm not sure
- [ X ] I observed this bug on a clean install of the OS
- [ X ] I have followed the project prerequisites
- [ X ] I have searched this repository for existing issues
- [ X ] I checked the FAQ and official documentation before creating this issue
- [ X ] I have read and understand the issue reporting guidelines 

## Bug description
Client like cellphone/laptop can connect to hotspot successfully, and even ping google.com/fackbook.com, etc. 
But curl and web browser can't get anything, and wifi status show "no internet connection"

## Your environment
1. Operating System: **Linux retropie 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux** 
2. Hardware and version: RPi 4B
3. RaspAP version: RaspAP v2.8.5
4. Clean install of a compatible operating system? Yes
5. RaspAP Quick Install or Manual setup?  Quick Install
6. Using default configuration? Yes
7. Simultaneous AP and managed mode? Yes
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP? qbitorrent-nox, clash

## Steps to reproduce

1.turn on Wifi client AP mode, with a local wifi connection of cource.
2.select wlan0 from interface droplist, and click start hotspot.
3.connect client to started hotspot.
4.Phone or laptop wifi menu show "no internet conneciton"
5.turn on terminal from phone or laptop, ping test to google.com/facebook.com, etc. successfully.
6.open web browser of those websites, all show no connection. "ERR_CONNECTION_CLOSED", "This site can't be reached".

## Screenshots

## Additional context

hotspot log looks ok:
Configuration file: /etc/hostapd/hostapd.conf
uap0: interface state UNINITIALIZED->COUNTRY_UPDATE
uap0: Could not connect to kernel driver
Using interface uap0 with hwaddr e4:5f:01:8f:cb:12 and ssid "PAiP"
uap0: interface state COUNTRY_UPDATE->ENABLED
uap0: AP-ENABLED

billz, 2022-08-17:
Per the [docs](https://docs.raspap.com/ap-sta/), this feature is experimental and unsupported. No hard feelings.

diegoxcn, 2022-08-17:
No worries, thanks.

Gitvaltiraheel, 2022-09-16:
No

closed 2023-12-12 by github-actions[bot]