RaspAP/raspap-webgui #1226: wlan0 to wlan1 hotspot

creed81, 2022-09-15:
## Checklist

- [x] This is a bug report
- [x] I observed this bug on a clean install of the OS
- [x] I have followed the project prerequisites
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

## Your environment
1. Operating System: RPi OS 32-bit (11 - bullseye with desktop)
2. Hardware and version: RPi 3B
3. RaspAP version: v2.8.5
4. Clean install of a compatible operating system? Yes
5. RaspAP Quick Install or Manual setup? Quick Install (`curl -sL https://install.raspap.com/ | bash -s -- --yes --openvpn 0 --adblock 1`)
6. Using default configuration? Yes, configured only from UI. As per [interface](https://docs.raspap.com/faq/#can-i-use-wlan0-and-wlan1-rather-than-eth0-for-my-ap) instructions 
7. Simultaneous AP and managed mode? No, but using internal and external wiki (Ralink MT7601U)
8. Onboard wireless chipset or external adapter? Both
9. Other software or services running with RaspAP? No clean install of just RPI OS and RaspAP

## Steps to reproduce
* Start with a clean RPI OS
* Install RaspAP using `curl -sL https://install.raspap.com/ | bash -s -- --yes --openvpn 0 --adblock 1`. 
* Confirmed interface with names wlan0 and wlan1 exist using `ifconfig`. wlan1 is autocreated after external USB wifi dongle is plugged in
* Try creating hotspot using RaspAP by selecting wlan1
* Find that hotspot is not created.
  * Also find that internet no longer works.
  * This can however be fixed by selecting wlan0 interface in UI and selecting DHCP, but not sure why this was changed.

## Screenshots

[image]
[image]

billz, 2022-09-15:
What are you reporting exactly? Please be specific in the bug description area.

creed81, 2022-09-15:
Updated

billz, 2022-09-15:
You must validate your external adapter before using it with this project.
Follow the steps in [this FAQ](https://docs.raspap.com/faq/#interfaces) and read our [issue policy](https://docs.raspap.com/issues/).

Your question is frequently asked on the [RaspAP subreddit](https://reddit.com/r/RaspAP). Look there for advice.

closed 2022-09-15 by billz

creed81, 2022-09-16:
I went through this FAQ and I had also mentioned this in the "Your environment" section of this issue. 
The adapter I am using is Ralink MT7601U which does support AP mode(present in the output for `iw`). I browsed other issues but they did not boil down to the issue I am facing but I can try the subreddit as well.