RaspAP/raspap-webgui #1231: Support for multiple Wireguard configurations/servers

neorrk, 2022-09-22:
I have multiple Wireguard servers that I want to choose from, but each time I have to upload a new .conf file. 

It would be nice if I could upload all of my configs and just choose which one I want to use from a menu.

billz, 2023-10-01:
@neorrk great suggestion :) https://docs.raspap.com/wireguard/#multiple-configs

Per our project's sponsorware model, this is available first to Insiders. Since you suggested it I'd be happy to give you free access to the Insiders repo, just say the word

neorrk, 2023-10-05:
@billz 
"Thank you so much for implementing this feature! I really appreciate it. I'd love to be part of the Insiders and try it out. Looking forward to testing it and providing feedback!"

closed 2023-11-08 by billz