RaspAP/raspap-webgui #1274: RaspAP fails to connect to wireless network with blank character at the end of the ESSID

realdeadbeef, 2022-12-28:
## Checklist

- [x] This is a bug report
- [x] I observed this bug on a clean install of the OS
- [x] I have followed the project prerequisites
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

I cannot connect to my home network from the WiFi Client page of RaspAP because the ESSID has a space at the end of it's name.

## Your environment
1. Operating System: RPi OS 32-bit Lite
2. Hardware and version:  Raspberry Pi 3 Model B+
3. RaspAP version: RaspAP v2.8.6
4. Clean install of a compatible operating system? Yes
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? Changed the DHCP range for the hotspot (wlan0)
7. Simultaneous AP and managed mode? I don't know
8. Onboard wireless chipset or external adapter? Onboard broadcasts network / USB Connects to existing network
9. Other software or services running with RaspAP? No

## Steps to reproduce
- Create a wifi network with the name "Nostromo "
- Attempt to connect to said network
- Observe that it does not connect due to a space at the end of the network ESSID

## Screenshots
[image]

## Additional context
I am able to connect to my phone's personal hotspot as it does not have a space at the end of the ESSID.

billz, 2022-12-28:
wpa_supplicant doesn't play nice with spaces. Consult [this source](https://wiki.archlinux.org/title/wpa_supplicant) or search "wpa_supplicant spaces" for help.

For reference, RaspAP executes the following:

https://github.com/RaspAP/raspap-webgui/blob/eb0d8bfa768155f6b92fc08939b2846d46cf9394/includes/configure_client.php#L20

If you can't connect to your SSID with this, neither can the UI.

realdeadbeef, 2022-12-28:
I can connect if I manually add it in wpa_supplicant.conf

billz, 2022-12-29:
I see what you mean. Output from wpa_cli scan is used to populate wpa_supplicant.conf. The method used did not capture the complete output as a string.

Trailing spaces in SSIDs are of course perfectly valid. eg.: 

```
$ sudo wpa_cli -i wlan0
wpa_cli v2.9
Copyright (c) 2004-2019, Jouni Malinen <j@w1.fi> and contributors

Interactive mode
> add_network
1
> set_network 0 ssid "POCO F3 "
OK
> set_network 0 psk "password"
OK
> enable_network 0
OK
<3>Trying to associate with SSID 'POCO F3 '
<3>Associated with ░░:░░:░░:░░:54:fd
<3>CTRL-EVENT-CONNECTED - Connection to ░░:░░:░░:░░:54:fd completed [id=0 id_str=]
<3>CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
<3>CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=US
``` 

I'll push a fix. Thanks for reporting.

PR #1275 (2022-12-30): Improve handling of wpa_cli output
Resolves #1274
closed 2022-12-30 by billz