RaspAP/raspap-webgui #1279: Please can the default IP address range for DHCP exclude the .255 address

stevef1uk, 2023-01-14:
I recently raised an issues which became a question as I wasted a lot of time trying to work out why a device attached to the WiFi wasn't working. This was because it was allocated the .255 IP address which isn't a valid one.
I looked at the code and there are only 3 files to change where the string '10.3.141.255' should be changed to '10.3.141.254'

modified:   README.md
modified:   config/090_wlan0.conf
modified:   config/defaults.json

Please can someone pick this up when they next make a functional change? It may save someone else from a lot of wasted time ;-)

I haven't built this project myself and don't have the time to learn how to do it and treat locally right now.

Thanks

Seve

billz, 2023-01-14:
https://docs.raspap.com/issues/

closed 2023-01-14 by billz

billz, 2023-01-14:
RaspAP conforms with the IANA Private Address Space as specified by [RFC 1918](https://www.rfc-editor.org/rfc/rfc1918).

> The Internet Assigned Numbers Authority (IANA) has reserved the following three blocks of the IP address space for private internets:
>```
>    10.0.0.0        -   10.255.255.255  (10/8 prefix)
>    172.16.0.0      -   172.31.255.255  (172.16/12 prefix)
>    192.168.0.0     -   192.168.255.255 (192.168/16 prefix)
>```

stevef1uk, 2023-01-14:
> RaspAP conforms with the IANA Private Address Space as specified by [RFC 1918](https://www.rfc-editor.org/rfc/rfc1918).

Completely true, but: https://www.techtarget.com/searchnetworking/answer/Can-you-assign-an-IP-address-ending-in-0-or-255