RaspAP/raspap-webgui #1288: Set RaspAP webserver domain path

fr3nchalp4a, 2023-01-24:
### Hardware :
- Raspberry Pi 4
- Raspberry OS 64bits booting from USB
### Software running on it :
- RaspAP (via Quick Installer)
- Docker with :
    - Portainer
    - Bitwarden (Vaultwarden/BitwardenRS exactly)
- Caddy

### Story
I have Bitwarden as a local password provider. I use RaspAP to extend a part of my WiFi, Portainer is just here to manage Bitwarden. Bitwarden needed HTTPS, so I used a trick with Caddy and DuckDNS to let Let's Encrypt give certs for a local IP (😆). 
Then I discovered the power of Caddy, and I wanted to have my 3 web services running under the same `example.duckdns.org/` URL, like `example.duckdns.org/bitwarden` to access Bitwarden, `example.duckdns.org/raspap` to access RaspAP, etc... This needed to tell the webserver that he's running with a path (i.e. `/raspap`). I did it with Bitwarden, but I didn't found any settings in RaspAP to tell lighttpd to run behind `/raspap`. Is there a way to do it, maybe with Caddy or something else if lighttpd can't do it ?

fr3nchalp4a, 2023-01-24:
For clarity :
I just want to know if there is a URL base parameter somewhere so I can set it and reverse-proxy in a subfolder : `example.duckdns.org/raspap`. It's just to skip the longer setup of Caddy to make it rewrite paths ([https://caddy.community/t/the-subfolder-problem-or-why-cant-i-reverse-proxy-my-app-into-a-subfolder/8575](https://caddy.community/t/the-subfolder-problem-or-why-cant-i-reverse-proxy-my-app-into-a-subfolder/8575)).

billz, 2023-01-25:
Take a look at `50-raspap-router.conf` used by lighttpd. 
More info https://docs.raspap.com/manual/#create-the-web-application

closed 2023-02-07 by fr3nchalp4a