RaspAP/raspap-webgui #1297: 404 errors on initial install

dmshimself, 2023-02-08:
I have a debian 11.6 minimum system and found that the bash script install gave a bunch of errors so followed the manual install and the only problem was not having curl installed and the version php.  After installing I can reach the admin web site and a dashboard displays, but all other pages give a 404 error.  Its likely something I'm doing wrong, but having done it twice with the same result, I'm not so sure 

Is there a recommended version of debian (or any other non-raspberry pi OS) which does 'just work' with the bash script method?  Or any thoughts on why I'm getting the 404 errors?

Any advice appreciated

billz, 2023-02-08:
https://docs.raspap.com/issues

closed 2023-02-08 by billz

dmshimself, 2023-02-08:
I'm taking a guess that you would like me to run the bash installation script after I have done the manual install.  If so, that actually makes things worse.  No errors were produced, but the previous raspap wireless network is no longer seen.  I've done a reboot and get the same thing.  Reverting to a snapshot restores the wirless network, but gives me the same 404 errors.

In that case,  to rephrase my original question - what version of debian x86/64 (11.0, through to 11.6) or Ubuntu (whatever version) is likely to work with the bash script installer as I would prefer to use it, but it didn't work with 11.6 and produce numerous errors.

If on the other hand my guess on what you'd like me to try is not to run the bash installer again, then I'm puzzled as I cannot see anything else that needs to be done, but some advice for my tired old eyes would be appreciated.