RaspAP/raspap-webgui #1316: 301 Moved Permanently error while installing

Toom4385, 2023-03-22:
I am unable to install

I get the below error
```
pi@raspberrypi:~ $ curl https://install.raspap.com | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   178  100   178    0     0    128      0  0:00:01  0:00:01 --:--:--   128
bash: line 1: html: No such file or directory
bash: line 2: syntax error near unexpected token `<'
'ash: line 2: `<head><title>301 Moved Permanently</title></head>
pi@raspberrypi:~ $
```

Thanks in advance

billz, 2023-03-22:
`curl -sL https://install.raspap.com | bash`

closed 2023-03-22 by billz

Toom4385, 2023-03-23:
When I used it it it nothing comes
```
pi@raspberrypi:~ $ curl -sL https://install.raspap.com | bash
pi@raspberrypi:~ $
``

groundwater-modeling, 2023-07-13:
the same problem. have you fixed it ?